stefandreyer/CODESYS-MQTT #2: more simple example

dkeipp, 2018-08-17:
can you provide a very simple example.
I think a clean raspberry project with just sending a message and subscribe to a topic would be great.
I would help with this but.. I don't get it to work because there are so many variables and structs.. I don't really see whats going on there and I can't get it to work in an other project.
Thank you

stefandreyer, 2018-08-17:
Hello dkeipp,

yes your are right, it is a little complex and maybe confusing. I'll make a smal test projekt to test.mosquitto.org.

but you must make sure that the dns clienent geht's your dns Server ip, for this take a look to commontypsandfunctions

BR

Stefam

dkeipp, 2018-08-17:
Thank you for the hint :-)
now I can send messages...
So maybe I can make a small example during the weekend..

for those who are trying to get this to work, here is the comment from GET_DNS_IP:
(don't forget to restart Runtime after that change)
`
for linux

change 
sudo nano /etc/CODESYSControl.cfg
or from runtime version 4.5.11.20
sudo nano /etc/CODESYSControl_User.cfg 
[SysProcess]
Command.0=shutdown

to 

[SysProcess]
Command.0=shutdown
Command.1=cat

for windows, needs powershell

"C:\ProgramData\CODESYS\CODESYSControlWinV3\???\CODESYSControl.cfg"
[SysProcess]
BasePriority=Realtime
Command=AllowAll
`

stefandreyer, 2018-08-17:
Hi dkeipp,

i am happy to heat that you can send Messages now.

And if you make this example i more happy....
i think it would be the best if you make it as an seperate projekt, not in the library. do you think to?

let me know if subscribe is working too.

BR Stefan

KevinRohn, 2018-08-18:
Hello folks,

the IP resolve functionality of your library seems very special to me. I added an example, which helps you to get the IP-Address resolved directly in IEC Code. 

I think it makes sense to add that function to your project :)

https://github.com/KevinRohn/cds-common-examples

stefandreyer, 2018-08-19:
Hello Kevin,

Looks gootd, i will have a try…

BR

stefandreyer, 2018-08-20:
Hi Kevin,

i implemnted it, just try it...

resolved a error in IP_CONTROL to.

BR Stefan

KevinRohn, 2018-08-20:
Hey Stefan,

I will check it later. Is there any chance to contact you via Mail?

You can contact me by visiting my private website: rohn(dot)io

stefandreyer, 2018-08-23:
Hello to all,

done an out of the box test projekt for the Raspberry, please test it...

BR Stefan

stefandreyer, 2018-09-24:
Everything fine?

I am going to close this issue.

BR Stefan

stefandreyer, 2018-09-24:
So here I am again,

now with new and tested Example Projects. Please have a look and tell your opinion.

BR Stefan

stefandreyer, 2018-09-26:
Ok,

I think now it is fine.

BR Stefan

closed 2018-09-26 by stefandreyer