stefandreyer/CODESYS-MQTT #3: Wago PLc not connecting to moquitto

YorickErnst, 2018-09-18:
Hi,

In an attempt to avoid paying 300 euro to Wago, I'm trying to implement your MQTT library on my PFC200.
I copied the code from your example into the codesys project and have all dependent libraries installed.

However when I run the code it is not connecting to the mosquitto ip.  Is this code dependant on a Pi specific implementation?

stefandreyer, 2018-09-19:
Hello Yorick,

i am happy that you found my projekt.
Id did this lib to avoid these 300 euro too… 

Ok I made a new TestOutOfTheBox fb. Please test it. I don't have any WAGO PLC to test it. Please make following test if it don't connecting.

Can your PLC accsess the internet? if not, change the value of the client input string to a reachable broker. Don't forget the port(:1883), you can use ipv4 address too.

If your PLC can connect to the internet, check if the resolving of the host name is working. To do this, go online an then browse to TestOutOfTheBox FB --> MQTTClient --> MQTTControl --> sResultIP. There you have to find your brroker ip. if not the dns resolving is not working so try fixed broker ip.

Please test it and give me your result. then we can look further.

BR

Stefan

stefandreyer, 2018-09-19:
Hello Yorick,
I made an update of CommonTypesAndFunction because the client didn't connect annymore on Windows host. Now it works again, maybe now by you too, giv it a try

BR

Stefan

YorickErnst, 2018-09-20:
Hi Stefan,

Thanks for the quick help there :)

Unfortunately it is still not working, although it's probably my fault, as your FB is working fine... 

Here is my code which is only slightly changed from your example:

`IF NOT init THEN
	init := TRUE;
	publish.SetMqttInOut(MQTT_IN_OUT:= ADR(MQTT_IN_OUT));
	subscribe.SetMqttInOut(MQTT_IN_OUT:= ADR(MQTT_IN_OUT));
	State.Init(
	State:= ADR(varSend), 
	Topic:= ADR('CODESYS-MQTT-BY-SD/State'), 
	StateName:= ADR(''), 
	StateValueOn:= ADR('On'), 
	StateValueOff:= ADR('Off'), 
	MQTT_IN_OUT:= ADR(MQTT_IN_OUT), 
	QoS:= SD_MQTT.QoS.ExactlyOnce, 
	RetainMqtt:= TRUE);
END_IF
timerSend(in := NOT varSend, PT := T#2S);
varSend := timerSend.Q; 
MQTTClient(
	MQTT_INFO:= MQTTInfo, 
	MQTT_IN_OUT:= MQTT_IN_OUT, 
	ENABLE:= TRUE, 
	URL:= '192.168.96.8:1883', 
	TIMEOUT:= T#5S, 
	TLS:= FALSE);
Payload:= TO_STRING(TICKS.GetTick(xDummy:= TRUE));

publish(
	Topic:= ADR('CODESYS-MQTT-BY-SD/Tick'), 
	PayloadString:= ADR(Payload), 
	PublishAsString:= TRUE, 
	QoSIn:= SD_MQTT.QoS.ExactlyOnce, 
	MRetain:= TRUE, 
	send:= varSend);
	
subscribe(
	Subscribe:= TRUE, 
	Topic:= ADR('CODESYS-MQTT-BY-SD/Tick'), 
	QoSSubscribe:= SD_MQTT.QoS.ExactlyOnce, 
	ExpectingString:= TRUE);

IF subscribe.Recived THEN
	ReciveSubscribePayload := subscribe.PayloadString^;
	ReciveSubscribeTopic := subscribe.TopicOut^;
END_IF

State();`

My PLC and the mosquito broker are in the same subnet and directly attached (not even a switch in between)

YorickErnst, 2018-09-20:
when I change the ip in my code by a the external test server it works fine. Is it possible that direct ip is not detected correctly?

stefandreyer, 2018-09-21:
Hmmmm….

I will check this ….

BR Stefan

stefandreyer, 2018-09-22:
Did you tried to use a Name for your local broker?

BR Stefan

stefandreyer, 2018-09-23:
Hi Yorick i figured out that there are some issues by using fixed ips.
I will resolve this ...

Thanks for that hint

BR Stefan

stefandreyer, 2018-09-23:
Hi Yorick,

please now take a look, I updates MQTT and CommonTypesAndFunctions.

BR Stefan

stefandreyer, 2018-09-24:
Hello again,

as I saw this mornig I, there are still some problems with static IPs. I will solve this today.

BR Stefan

stefandreyer, 2018-09-24:
Hi Yorick,

its done, please test it. The Library now should work in an full static environment.

BR Stefan

YorickErnst, 2018-09-24:
Awesome work Stefan.

I just got back from a trip to Norway and now my PLC can connect to the MQTT broker.
Thanks very much that.

Kind regards,
Yorick

closed 2018-09-24 by YorickErnst