stefandreyer/CODESYS-MQTT #6: Any chance for a port to Phoenix Contact

Portisch, 2018-10-09:
Is there a way to convert this project to be able to use in a Phoenix Contact PLC?
Or can you give us the raw ST code of the library?

stefandreyer, 2018-10-11:
Hi Portisch,

the raw ST code is inside the projekt. You can use 
[https://store.codesys.com/engineering/codesys.html](url) (free, just register)
to have a look.

I can provide an OPENplc xml. Is PC Works able to import this?

BR Stefan

stefandreyer, 2018-10-12:
Hi Portisch

i published an OPENplc XML. Can you use ist?

BR Stefan

Portisch, 2018-10-15:
Where I can find the OPENplc XML?
PC Worx look like able to import PLCopen XML files.

Portisch, 2018-10-15:
PC Worx is able to import PLCopen XML - but it isn't working.
Anyway I found another way to make a bridge between PC Worx and MQTT.

closed 2018-10-15 by Portisch