stefandreyer/CODESYS-MQTT #7: best approach to use MQTTPublish

MichielVanwelsenaere, 2018-10-28:
Hi Stefan,

first of all; very nice work, it's great to see an open source library with advanced functionality for Codesys mqtt!

I currently have a program where multiple FB's publish updates to a queue and I'm now looking in transmitting the updates to an mqtt broker one by one.

I've checked your docs on the FB's and the examples used, in your examples your using a timer to create a positive edge to trigger a send each two seconds yet the `MQTTPublish` FB had an output `done` to indicate when a successful transmit has occurred. 

Now my question: is it sufficient to verify if `done` is high to start a new mqtt publish? What would be the best way to publish messages at a high rate at QOS2? (I read throughput of 120 messages/second here somewhere and was wondering how they are achieved )

stefandreyer, 2018-10-28:
Hi Michiel,
you have found docs? Show me where… ;)
OK, for my MQTT client you don't need a queue, just use one `MQTTPublish`FB for each Value, String, State ...
Take a loot to my `MQTTValue` FB, it is fine for updating values to a broker.

You have to check `done`, because it shows you that the Payload an the topic can be changed again, until it is not set you must not change Payload, Topic!

The `HANDLE_MQTT` FB ist able to work on 50 packets to the broker at one time. So if 30 Values are changing in one cycle this 30 publish packets will be in the next tcp packet(Limit by 2k Byte). If this is not  sufficient for you, you have to increase `QoS_BUFFER_SIZE` in the `HANDLE_MQTT` FB.

Take a look in the `TestMqtt` there I am using a array of publish FB's for the stress testing, with it an my optimations I can publish up to 240 packets per second on QoS 2! :)

so Michiel, test it and have fun!

BR Stefan

stefandreyer, 2018-10-29:
Hi Michiel,

yes you are Right, this docs are build automaticaly by CODESYS, but I don't know if they are still actuall.

Do you mean the Timeout Output? If this is going to be high, this is just a info for you. the `MQTTPublish` FB retrys transmitting if the connection is established again. You just have to wait untill send is set.

Retrigering by user is not necessary.

Please let me know if it works like you want it.

BR Stefan

MichielVanwelsenaere, 2018-10-30:
Hi Stefan,

that's great, thanks!
Works how I'd like it to work (no need to worry about transient network issues)

Best regards,
Michiel

MichielVanwelsenaere, 2018-11-01:
Stefan,

a problem I noticed is that the `MQTTPublish` output value `done` starts as false. I'm attempting to use it to determine if that FB (it's an array like in your tests) is done sending and can start sending a new value. Yet it has a init value of false which blocks me a bit, I resolved it by using the condition below:
`MqttVariables.publish[iloopcounterpublish].PayloadLength = 0) OR MqttVariables.publish[iloopcounterpublish].done THEN`

yet this feels dirty.. Is there a better way to approach this or should the init value of the `done` output by true instead of false?

additionally I'm having trouble finding what's the best way of using the `send` input to trigger a publish. I have an array of 50 publish elements and only want to have a publish occur once ofcourse, any advice on how to approach this? Do I need 50 ton timers? won't the new positive edge of the timer trigger a resend? 

Best regards,
Michiel

stefandreyer, 2018-11-01:
Hi Michel,

that `done`  is set to `false` at the start of the FB is correct, so I clear the done bit. The `done` Output is later set to one in the FB if the HANDLE_MQTT FB quits the sucsessfull sending of the publish packet.
So this is a Kind of making an Edge for one PLC cycle.

And you can't read `MqttVariables.publish[iloopcounterpublish].PayloadLength` to determinant that sending was sucsessfull, because is only read by the handle FB.

A new Edge won't retrigger the FB if he is already sending, but be Aware if this Edge create a new payload content with different length, then your publish data will be corruptet.

Please take a look in my `MQTTValue` FB this is a good example how to use the `done` output and creating the publish Content. And there is a TON for every instance of `MQTTValue` so you can define in which time and value intervall a value have to be transmitted.

The big Benefit of this FB is that everything wich he needs is set at init and then you have just to call the instance with `MQTTVALUE();` and the FB is handling everything. See `TestPublishValue` FB too.

I have a FB for one bit values too, 'MQTTState'.

What kind of data do you transmitt? Maybe it makes sense to create a FB for this purpose too.

BR

Stefan

MichielVanwelsenaere, 2018-11-01:
I'm sending statechanges on my outputs but also events on my inputs: single, double, long presses on pushbuttons. I wan't to keep things generic as I might add more stuff in the future (it's a home automation project). From your `MQTTValue` implementation I can see you're addressing a variable directly to publish it through mqtt which looks handy but is it scalable (what if I have hundreds of values?) 

My approach to tackle the problem and keep things generic is by using a queue in between. The queue is a global variable to which all FB's can publish (each with their own unique topic and payload). I have another task with 50 Publish FB's to process the load in the queue. 

anyway, the `MqttValue` should be sufficient as example for me to build a new FB to do the job. Thanks again for your help!

stefandreyer, 2018-11-02:
Hi Michiel,

You can use the MQTT_IN_OUT struct a global variable too. 

But your queue is a nice approache too. Maybe you can publish your queue library too, I would like to have al look.

Have sucsess!

BR Stefan

MichielVanwelsenaere, 2018-12-16:
Hi Stefan,

I'm having some trouble to get this to work. I have no idea why but my new FB for publishing MQTT values from a queue isn't publishing anything (while it's very similair to your `MQTTValue` FB). I'm wondering if you have some time/interest to have a look?

Best regards,
Michiel

stefandreyer, 2018-12-17:
Hi Michiel,

I would like to have a look. Should be not to difficult to solve. How you would like to transmitt your library?

BR

Stefan

MichielVanwelsenaere, 2018-12-17:
Great! I also think it's something small but pinpointing it is a pain. I also saw om mqtt broker that the PLC isn't connecting. but that's perhaps because my Publish isn't working as it should. I prefer not to upload a link over here. Do you have a mail address you can share over here?

stefandreyer, 2018-12-17:
Hi Michiel,

send me your email to 

kellum.vidyuth@plutofox.com

i will then send you a mail with my right adress.

BR Stefan

stefandreyer, 2018-12-17:
Hi Michiel,

please write your Email in the Email, this Provider doesn't show me the Sender.

BR Stefajn

stefandreyer, 2018-12-17:
Hi,

you should have a mail, Maybe spam?

MichielVanwelsenaere, 2019-03-25:
closing this one, thanks for your help!

closed 2019-03-25 by MichielVanwelsenaere