stefandreyer/CODESYS-MQTT #8: noTLS example does not work

bmshine, 2018-10-30:
Hallo Stefan,

ich habe mal deinen MQTT Baustein geladen und den Beispielprogramm ausprobiert. Leider funktioniert nur die TLS Variante.
Die ohneTLS Variante verbindet sich weder auf test.mosquitto.org noch meinen lokalen MQTT server.
Muss ich bestimmte Versionen von den LIB's laden oder funktioniert es auch mit den neusten?
Aktuell verwende ich Codesys 3.5 SP13 Patch2.

Gruss Markus

stefandreyer, 2018-10-30:
Hallo Markus,

auf welchem System läuft die Runtime?
Funktionieren muss dass alles mit den Bibliotheken die du in meinen Repositories findest.

Grüße Stefan

bmshine, 2018-10-31:
Hallo Stefan,
auf Windows10 funktionierts nicht. Auf dem Raspi mit lokalem MQTT Server funktioniert das Beispiel. Mit test.mosquitto.org als Server funktionierts da leider auch nicht.
Gruss Markus

stefandreyer, 2018-10-31:
Hallo Markus,

kann das Problem nachvollziehen, gehen dem nach. Grüße

stefandreyer, 2018-10-31:
Hallo Markus,

gestern hatte ich die selben Probleme wie du beschrieben hast. Heute Klappt die Verbindung zu test.mosquitto.org. Ich denke die werden da was versucht haben.

Bitte versuche es nochmal und geb mir Rückmeldung, vielen Dank

Grüße Stefan

bmshine, 2018-11-04:
Hallo,
das Windows Beispiel funktioniert bei mir immer noch nicht. Die Raspi Variante mit lokalem MQTT Server funktioniert super. Vielen Dank.
Gruss Markus

stefandreyer, 2018-11-04:
Hallo Markus,

schau mal bitte nach folgenden Variablen:
MQTT_CONTROL : step
MQTT_CONTROL : SResultIP  (sollte ip von test.mosquitto.org enthalten, wenn nicht stimmt was mit dem DNS nicht)
IP_CONTROL_TLS : State

und geb mir die Aktualwerte.

Aktuell funktioniert das beispiel bei mir.

Grüße Stefan

kurta241, 2018-12-16:
Hallo Stefan,

Das Projekt "TestMQTTGithubWindowsWithoutTLS.project" funktioniert bei mir leider auch nicht. 
MQTT_CONTROL : step = 30
MQTT_CONTROL : SResultIP  = 3.121.19.92  (Adresse von broker.hivemq.com)
 IP_CONTROL_TLS : State = konnte ich nicht finden
Mit dem Broker test.mosquitto.org konnte ich mich auch mit einem MQTT-Client auf dem PC nicht verbinden, deshalb habe ich broker.hivemq.com verwendet. 

Mit Wireshark sehe ich den TCP-Verbindungsaufbau (3-Way-Handshake) und 10s später den Verbindungsabbau. Dazwischen findet keine Datenübertragung statt.

Was kann ich noch zur Fehlersuche tun?

SG
Kurt

stefandreyer, 2018-12-17:
Hallo Kurt,

habe das Beispiel heut früh mit hivemq.com getestet. Lief bei mir. Es sieht so aus als wenn er nicht mit dem Verbindungsaufbau fertig wird. Du musst in dem super() Aufruf in IP_control_TLS online öffnen und dort nach dem Städte schauen.

Bei test.mosquitto.org spielen sie ab und zu rum. Vielleicht Lags daran. Hatte ich auch schon.

Grüße Stefan

kurta241, 2018-12-17:
Hallo Stefan,

Danke für den raschen Hinweis. 
TC_Connect bleibt im state=52 hängen. SysSock2Connect liefert einmal als Ergebnis pResult=518, dann immer pResult=525 (socket is already connected). Da nur bei pResult=0 zum nächsten Schritt (250) weitergeschaltet würde, geht nichts mehr weiter.

Ich habe nun das Testbeispiel für den Raspberry ausprobiert - funktioniert problemlos. Ich möchte die MQTT Bibliothek sowieso auf dem Raspberry einsetzen und muss daher den Fehler im Windows-Beispiel nicht unbedingt finden, helfe dir aber gerne bei der Fehlersuche.

Danke, dass du diese MQTT-Bibliothek zur Verfügung stellst!

Schöne Grüße
Kurt

stefandreyer, 2018-12-18:
Hallo Kurt,

welche Windows Version nutzen du?

Grüße Stefan

kurta241, 2018-12-18:
Hallo Stefan,

Auf meinem PC ist Windows 10 Pro Build 1803 installiert.

Schöne Grüße
Kurt

kurta241, 2018-12-19:
Hallo Stefan,

Bei mir sind folgende Versionen installiert:
SysSocket2, 3.5.10.0 (System)
SysSocket, 3.5.13.0 (System)

C:\netstat -an
zeigt eine hergestellte TCP Verbindung auf port 1883, und mehr als 10 wartende. Die Verbindung wird hergestellt und wieder verworfen. TC_Connect bekommt das aber nicht mit, da SysSock2Connect nie pResult=0 liefert. SysSock2Connect liefert 525 und diese Rückgabewert wird im CASE ab Zeile 388 nicht berücksichtigt.

Schöne Grüße
Kurt

kurta241, 2018-12-19:
Hallo Stefan,

Habe jetzt SysSocket2, 3.5.13.20 Bibliothek hinzugefügt. In CommonTypsFunctions steht immer noch die SysSocket2, 3.5.10.0. Ist hier etwas mit den Versionen durcheinander?

Schöne Grüße
Kurt

Tommilein, 2019-01-18:
Hallo Stefan,
Habe das Testprogramm(ohne TLS) zum Laufen gebracht. Ich kann sehen, das die MQTT-Übertragung zu 'test.mosquitto.org:1883' sttfindet. Jedenfalls kommen im mqtt.fx Daten an.

Aber nun verzeihe einem nicht ganz so gewieften Codesys-Programierer. Wie kann ich nun das Testprogramm für meine eigenen Programme nutzen. Ich erkenne nur eine Eingabe-Variable für die URL. Ich müsste aber auch noch einen EIngabewert zum "publish" und eine Ausgabevariable für "subscribe" haben, damit ich was geordnet übertragen bekomme. Ein bissel "status" wäre auch ganz schön.

Ist diese Test-Programm ohen jeden Kommentar und mit einer Programmzeile von vornherein so konzipiert, oder finde ich wieder mal nicht alles...

Für ein paar kleine Hinweise wäre ich dankbar.

Übrigens mit einem lokalen Broker konnte ich eigene Daten hin und herschicken (forcen für publish und nur Anschauen für subscribe, verarbeiten konnte ich nichts.

Mfg
Thomas

Tommilein, 2019-01-19:
Hallo Stefan, 

Meine Antwort begann vor einer Stunde so:
     _vielen Dank für die schnelle Antwort. Da komme ich vielleicht weiter. Solche Tricks wie 

      **_„…testMQTT:SD_MQTT.TestOutOfTheBoxNoTLS; Symbol suchen --> gehe zur Definition“ …“_**

     sind für mich eben nicht Standard – man lernt immer dazu. 
     Ich versuche nun den Code aus TestOutOfTheBoxNoTLS
     in die „POU“ zu übertragen, um dann dort die Variablen füllen zu können.
     Aber schon wenn ich nur  die Variablen übertrage, kommen massenweise
     die Compilerhinweise auf Fehler._

Dann habe ich immer weiter geforscht, da ich nicht verstehen konnte (und wollte), dass ich so beschränkt sein soll. Und siehe da, wenn man nur stur genug ist, kommt man zum Erfolg:
[image]

 
Aber ich gebe zu: ohne Deinen Hinweis hätte ich noch ewig rumgesucht !!!

Ist eben schön, wenn das Internet nicht nur für Selfies gut ist. 
Jetzt kann mein Projekt weitergehen,  noch mal vielen Dank für Deinen Hinweis. Aber noch mehr Respekt für Deine Library !!   

Viele Grüße
Thomas

stefandreyer, 2019-01-19:
Hallo Thomas,

schön das du voran kommst.

Aber sei gewarnt, ich aktuallisiere gerade die Library und stelle das subscription Konzept um. Ich verwende nun Callbacks. Pro: viel schneller und sicherer, Kontra: vielleicht nicht ganz so einfach. 
Ich baue aber ein paar schöne Beispiele dazu.

Grüße Stefan

Tommilein, 2019-01-19:
Hallo Stefan,

 

ist doch toll, wenn Du die Library weiterentwickelst (nicht : fire'n'forget ) .

 

Wenn Du aber u.U. gerade am Computer bist – kannst Du mir mal einen Hinweis zu dem öffentlichen Broker geben ?

 

Wie komme ich denn zu einem „eigenen“ Topic. Ich möchte getrennte Topics für Senden und Empfangen anlegen. Aber mit meinen Topics geschieht nichts – kein Senden/Empfangen.

 

Sind die schon verwendet oder wie komme zu eigenen getrennten Topics (auch für state).

 

Mit freundlichen Grüßen

Thomas

Tommilein, 2019-01-19:
(UPS, habe direkt auf obige Nachricht geantwortet, ist besser, ich schreibe hier alles rein.)

Hallo Stefan,

ist doch toll, wenn Du die Library weiterentwickelst (nicht : fire'n'forget ) .

Wenn Du aber u.U. gerade am Computer bist – kannst Du mir mal einen Hinweis zu dem öffentlichen Broker geben ?

Wie komme ich denn zu einem „eigenen“ Topic. Ich möchte getrennte Topics für Senden und Empfangen anlegen. Aber mit meinen Topics geschieht nichts – kein Senden/Empfangen.

Sind die schon verwendet oder wie komme zu eigenen getrennten Topics (auch für state).

Mit freundlichen Grüßen
Thomas

stefandreyer, 2019-01-19:
Hallo Thomas,

Ein eigenes Topic? Nun die öffentlichen Broker haben was von Anarchie. Du kannst zu jeden Vorhandenen Topic publishen oder einfach ein neues aufmachen. Es gibt für Topics bei Brokern kein "Anmelden", "Registrieren" oder "Reservieren". Einfach machen. Aber so ist es auch anders herum. jeder kann deine Topics lesen und auch darauf schreiben.

Wenn du im Codesys ein Publish senden willst, mache dafür eine eigene Instans des MQTTPublish FBs auf und vergess das Initialisieren nicht. Genauso für die Subscribtion.

Grüße Stefan

Tommilein, 2019-01-22:
Hallo Stefan, 
bin mit Deinen Hinweisen eigentlich gut zurechtgekommen - bisher im Testprogramm. Wenn ich nun die MQTT-Teile in meine Anwendung kopiere, da knalle ich gegen die oscat-network 1.2.1.  Durch FTP-Aufruf fahre ich voll gegen die Wand (Totalabsturz Raspberry). Nun finde ich auf den Oscat-Seiten nichts mehr von network 1.3.0. - auch keine Beschreibung, wo ich die evtl. vorhandenen Aufruf-Unterschiede nachvollziehen könnte. 
Hast Du eine Beschreibung ? -- oder was ist da bei oscat los. 

Viele Grüße
Thomas

stefandreyer, 2019-01-23:
Hallo Thomas,

ich habe den IP_Controll anpassen müssen, hast du meinen aus der CommonTypesAndFunction lib?Ich habe nich kontrolliert ob 2 verschiedene FBs kollidieren. Versuch mal auf meine Version zu gehen. Ich habe auch eine OSCAT Network lib im repository, vielleicht geht die für dich.

Grüße Stefan

Tommilein, 2019-02-02:
Hallo Stefan,
brauche nochmal Deine Hilfe. Wie ist es angedacht, wenn nun mehrere Mqtt-Clients auf die Codesys-Maschine einstürmen. Kann ich alle Topics "irgendwie" über einen einzigen Subscribe-Aufruf übernehmen und danach im Codesys analysieren, wo das Telegramm herkommt. Oder muss ich pro topic je eine subscribe-Instanz definieren ??
Ich will die Welt nicht nochmal erfinden...
Danke und viele Grüße
Thomas

Tommilein, 2019-03-10:
Hallo Stefan,
ich muss Dich nochmal bemühen. 
Bei mir geht seit 8.3.2019 der Broker "test.mosquitto.org"  nicht mehr. Sowohl das Tool mqtt.fx connected nicht mehr als auch ein Arduino, der lange Zeit vor sich hin sendete, steigt aus. Der Arduino geht sofort wieder mit einem anderen Broker. Auch Dein neuestes Programm connected nicht.
Nun sind die öffentlichen Broker auch im Betrieb wahrscheinlich nicht sicher und ich hatte die Idee, den broker zu wechseln wenn einer nicht geht. Der Arduino macht das auch - er wechselt enfach, wenn er merkt es geht nichts raus.
Der Raspberry weiß aber davon u.U. nichts, bzw. weiß er nicht wohin der Arduino gewechselt hat. Die Lösung ist aus meiner Sicht die: der Raspberry muss alle vorgesehenen Broker permanent abfragen (vielleicht jede Minute) und wenn er von dort was sinnvolles bekommt, dann connected er sich dahin mit der richtigen Abfrage-Geschwindigkeit.
Meine Frage: Mit Deinem Programm muss ich dann eine komplette zweite(dritte) Instanz mit einem anderen Broker anlegen (??) und das entsprechend abfragen - oder gibt es innerhalb des FB noch eine Eingriffsmöglichkeit ??

Vielen Dank
Thomas

stefandreyer, 2019-03-10:
Hallo Thomas,

habe es gerade selber Probiert, geht nicht, bestimmt spielen die übers WE...
broker.hivemq.com funktioniert.
Ja, öffentlichen Broker würde ich da auch nicht über den weg trauen.
habe vor kurzem davon gehöhrt: https://www.cloudmqtt.com/ gibt da ne kostenlose Variante, ist vielleicht was für dich.
Ansonnsten, ja du musst den Clienten 2 mal Aufrufen. Das ganze wird aber ein Problem, da ich dieses Broker hopping nicht vorgesehen habe. Du müsstest die Variablen alle Doppelt zuweisen... hmmm oder den MQTT_IN_OUT umschalten, je nachdem welcher Client verbunden ist. Dann weißt du aber immer noch nicht mit welchem der Arduino verbunden ist. Für mich klingt das alles sehr unsicher.
Wenn dein Codesys auf nem Raspi läuft könntest du dir da einen eigenen Broker aufsetzten und ins Netz lassen(DynDns), würde auch gehen, natürlich nur verschlüsselt.

Grüße

stefandreyer, 2019-03-11:
Hallo Thomas,

heut geht's wieder....

Grüße Stefan

Tommilein, 2019-03-12:
Hallo Stefan, 
vielen Dank für den Hinweis. Ich habe noch zwei gut funktionierende Broker gefunden:
 - mqtt.fluux.io
 - broker.mqtt-dashboard.com
wenn dann noch "test.mosquitto.org" dazukommt wird wohl einer von den dreien funktionieren - immer vorausgesetzt, ich bekomme das gleichzeitige Abfragen der 3 Broker im Codesys hin.

Publish von Codesys geht mit 2 Brokern.

Ein erster Versuch mit 2 Brokern zu "suscriben" hat nur kurz funktioniert,dann geben sie den Geist auf.
(mein - wahrscheinlich amatuerhafter - Glaube an unabhängige Instanzen ist damit stark erschüttert...)

Mein Gedanke ist nun der:
ich muss abwechslend die Broker abfragen. Ist was angekommen, wechsle ich danach den Broker
Problem dabei ist: kommt nichts, wechsle ich auch nie den Broker, da ich mich auf das "subscribe.Recived"-Signal berufe.
Ich brauche also ein Signal, das "nichts" empfangen wurde. 
Kann ich in Deinen Programmen sowas finden?

Das Ganze ist jetzt zwar von hinten durch die Brust: aber für jede andere Idee bin ich dankbar.
Viele Grüße
Thomas

stefandreyer, 2019-03-13:
Hallo Thomas,

es sollte vielleicht mit einer Instanz des Clienten gehen, du musst nur den Broker string für die Verbindung(url) wechseln und dann verbinden. Dann muss dein Arduino noch zyklisch eine lebensmeldung an eine bestimmte Topic senden, und schon kannst du den vom Arduino gewählten broker herrausfinden.

Verbinden, Lebensmeldung abwarten --> timeout, trennen, url wechseln, verbinden.

Grüße Stefan

Tommilein, 2019-03-14:
Hallo Stefan,
ich bleibe mal vorläufig bei 2 Instanzen. Das funktioniert hinreichend gut. Bei einer Instanz habe ich das Problem mit dem "Lebenszeichen". Der Arduino soll ja solange wie möglich im "deepsleep" bleiben. Das sind durchaus mal 10 Minuten. Da würde das Erkennen des fehlenden Brokers sehr lange dauern. Aber ich denke, das ist  "Programmierphilosophie". Ich habe mit Deiner Hilfe auf jeden Fall eine Lösung gefunden.

Nun aber gehen meine Probleme weiter. Ich möchte jetzt mit TLS arbeiten. Dein Beispielprogramm hat sofort funktioniert. ABER: das kam mir etwas einfach vor. Ist da jetzt irgendwas verschlüsselt oder ist ein Passwort versteckt, oder, oder ...
Kannst Du mir da noch eine paar Hinweise geben. Im Netz gibt es viele Infos, aber wie ich diese in Dein Programm integrieren kann - rätselhaft.

Vielen Dank (und ich hoffe, ich gehe Dir nicht zu sehr auf den Geist)
Gruß Thomas

Zitat:
_Security gehört zu den wichtigsten und gleichzeitig am stärksten vernachlässigten Themen im Internet der Dinge._ 
und:
_Applikationsentwickler sind nicht immer Sicherheitsexperten und sollen sich auf die Geschäftslogik ihrer IoT-Anwendung (Internet of Things) konzentrieren, sei es eine mobile App, Software oder Hardware._ 

.. wie war !!!!

Tommilein, 2019-03-14:
Hallo Stefan,
gleich nochmal hinterher:
Dein Programm mit Port 8883 sendete etwas und ich konnte es mit dem Tool mqtt.fx mit Port 1883 empfangen, deshalb ging das so schnell. Nun habe ich am mqtt.fx rumgesucht und eine ganze Menge Einstellungen gefunden, die nicht so recht einfach sind:
[image]
Hier habe ich noch keinen Connect zum Server zustandegebracht, wenn ich den kleinen Haken bei "enable SSL/TLS" setze.

Im Internet habe ich im Zusammenhang mit TLS sowas wie 
         _**espClient.verify(mqtt_fprint, mqtt_server)**_
gefunden. Wo in Deinem Programm wäre das zu konfigurieren? Ich brauche doch einen Schlüssel oder ein Schlüsselfile.

Also wie schon gesagt - Hiiilfe !!

Grüße
Thomas

stefandreyer, 2019-03-14:
Hallo Thomas,

willkommen in der Gruppe "Bedänkenträger IoT". Unsere Gruppe hatte dieses Jahr schon großen Zuwachs. Wir sind jetzt zu zweit!

Solltest du dir ein Sprachassitens fähiges Gerät großer IT Konzerne ins Haus stellen, wird dir die Mitgliedschaft gekündigt!

Spaß beiseite.

Schau mal hier:
https://forum-de.codesys.com/viewtopic.php?f=10&t=5716
Da gings um die Entwicklung zu dem Ganzen. Du kannst den TLS Kontext selber definieren mit:

TLS_CONFIG:CommonTypesAndFunctions.TLS_CONFIG;

und übergabe mit:
MQTT_IN_OUT.TLS_CONFIG := ADR(TLS_CONFIG);

Aber in Wirklichkeit verschlüssel ich nix, gebe es nur vor :D

Schau die die Kommunikation im Wireshark an da siehst du den Handshake.

Die Kommunikation wird verschlüssel, sollte ein Passwort vom Server angefordert werden ist dies Verschlüsselt. Es wird nicht geprüft ob der Server der Richtig ist, sollte jemand die DNS Kommunikation korummpieren kannst du dich mit einem falschen Server verbinden.

Server Zertifikaten können Implementiert werden, habe ich noch nicht.

MQTT FX scheint TLS V2 nicht zu unterstützen ....

Genau, ich Sende verschlüsselt. Wenn der broker dazu noch unverschlüsselt akzeptiert, dann ist es so.

Kann der Arduino TLS?

Codesys nutzt hier OpenSSL Da kannst du dir mit der Doku eine eigene Chiper List zusammen bauen.

Viel Erfolg!

Und geh mir ruhig auf den Keks, so frische ich auf und lerne dazu!

Grüße Stefan

Tommilein, 2019-03-15:
**PROBLEM !!!!    PROBLEM !!!!   PROBLEM !!!!**
Hallo Stefan,
ich habe ein Problem, mehr als 80 Zeichen zu empfangen. Ganz typisch wäre eine Definition eines Strings ohne Länge. Im Programm kann ich aber einfach nichts finden, wo so ein zu kurzer String wäre:
[image]

Irgendwo in den Variablen habe ich eine für mich komische Zahl entdeckt, ich kann sie aber nicht deuten.
[image]

Kannst Du mal schauen, ob Du da nicht einen kleinen Fehler irgendwo hast, damit ich danach ohne Zweifel bei mir weitersuchen kann, wobei die Länge schon bei "ReciveSubscribePayload" fehlt - obwohl dort STRING(255) definiert ist und das mit einer Pointer-Operation gefüllt wird (??)
[image]

Mir ist aufgefallen in  **"FUNCTION_BLOCK MQTTSubscribe EXTENDS MQTTBase"** 
das dort definiert ist:

**_PayloadString:POINTER TO STRING;_**

Kann es sein, das dann Codesys bei der Operation

**_ReciveSubscribePayload := subscribe.PayloadString^;_**     

automatisch bei 80 Zeichen aufhört ????
Könnte die Definition 
**_PayloadString:POINTER TO STRING(255) ;_**
das Problem beheben. Ich habe nicht viel Ahnung von Pointer-Operationen (leider).

Schau doch bitte mal...
Grüße
Thomas

Tommilein, 2019-03-15:
Update: Es scheint wirklich so zu sein, das die erwähnte Pointer-Operation nur 80 Byte rüberholt:

[image]

Jetzt wäre alles gut, wenn ich nicht den "Vogel" hätte, nun doch lieber reine Byte-Arrays zu übertragen. Denn mit Strings kann ich nun mal keine 0-Bytes senden. Da ist immer Schluß.

Is eben so...
Viele Grüße
Thomas

stefandreyer, 2019-03-15:
Hallo Thomas,

ich kenne das Problem, habe es nur noch nicht im ListStringCallback bedacht.

Der Singel String Callback macht das mit memmove und ignoriert die String Grenzen knallhart, da gehen auch mehr wie 255 byte....

Ziehe ich noch glatt.

stefandreyer, 2019-03-15:
Zu dem TLS Thema schreib ich nochmal...

stefandreyer, 2019-03-16:
Hallo Thomas,

habe eine neue Version hoch geladen, sollte nun gehen.

Das mit den String Pointer ist sehr frustrierend. Wie du schon festgestellt hast, kann man auch bei den Pointern eine länge festlegen. Dann kann man mit den Pointern auch String > 255 zeichen Kopieren.

Sollte nun aber ein `p1:Pointer to String(1000)` auf String(80) zeigen und man kopiert nun über den Pointer einen String > 80 dann überschreibt codesys knallhart follgenden Speicher mit ungeahnten konzequenzen. Ich habs gerade Probiert, hat schön gekracht...

Also, werde ich bei den Subscriptions noch die Längen der Ziele mitgeben. Sonst wird mal ein zu langes Payload abgschickt und die Steuerung geht hops ....

Grüße Stefan

Tommilein, 2019-03-17:
Hallo Stefan,
glaubst gar nicht, wie stolz ich auf mich bin, daß ich mit meiner Vermutung richtig lag. Im Normalfalle bin ich nie sicher, ob irgendeine Ungereimtheit nicht die Ursache in meinen Programmzeilen hat - meistens nämlich...

Das mit TLS wäre eine schöne(wichtige) Geschichte, wenn Du sie so aufbereiten/beschreiben könntest, das der Normalsterbliche die Sache nutzen kann. 

Danke für Deine Arbeit.
Grüße
Thomas

Tommilein, 2019-03-25:
Hallo Stefan,
hab wieder mal ne Frage:
wie komme ich denn an die Endemitteilung beim "publish" ran. In "SD_MQTT.MQTTState" sieht es ja so aus, als gibt es ein Flag "SendDone". Wie komme ich denn an das Flag heran ??
Kann sein, das das Grundwissen von Codesys ist, aber ich weiß es nun mal nicht.

Vielen Dank
Thomas

stefandreyer, 2019-03-25:
Hallo Thomas,

das ist ein Ausgang am FB und kommt für genau einen Zyklus wenn das Paket vollständig abgearbeitet wurde. Dort heißt es "done"

Grüße Stefan

Tommilein, 2019-03-25:
Danke !!

 

Grüße Thomas

Tommilein, 2019-07-11:
Hallo Stefan,

habe nun endlich die neue Bibliothek runtergeladen und mit viel Mühe geupdated (ich mach das nun mal nicht jeden Tag und die Bibliotheks-Versionen sehen nach aussen hin alle gleich aus. Ehe man dann mitbekommt, das man die alte immer wieder verwendet - Katastrophe).
Leider läßt sich das Beispielprogramm nicht fertigkompilieren, da ich immer den Fehler

[image]

Fehler:
**_[FEHLER]        cmperrors interfaces, 3.5.5.0 (system): C0338:  Die Bibliothek 'cmperrors interfaces, 3.5.5.0 (system)' wird nur in 32-Bit-Applikationen unterstützt
                Kompilierung abgeschlossen -- 1 Fehler, 0 Warnungen_**

bekomme. Ich fahre **_Codesys 3.15 SP14 + 32bit_**   . Wie kann ich denn das Problem ausmerzen ??
Allerdings  habe ich Windows 64bit, was wohl die Ursache ist.

Als workaround kann ich mir die Umstellung auf den raspberry vorstellen. Ich kriege aber einfach das Gerät "**_Codesys Control for Raspberry Pi SL_**"  nicht in das Testprogramm...

Viele Grüße
Thomas

stefandreyer, 2019-07-11:
Hallo Thomas,

bei mir in der Bibliothek sieht es so aus :

[image]

Hast du wirklich alle Bibliotheken aktuell?

Bei mir läuft Codesys V3.5 SP14 + 64 bit…

Ich glaube ich hatte auch mal ein 32 bit codesys installiert und dann damit auch Probleme(dem Codesys).

Aber für die Runtime müssten 32 und 64 bit funktionieren.

Und ich glaube mittlerweile haben nur noch sehr wenige Entwicklungs PC 32 bit Betriebssysteme.

Mach doch einfach mal ein neues Projekt, dort einen Raspi einfügen und dann den Test FB aufrufen.

Mal schauen ob das klappt.

Ich hab dich noch gar nicht auf der Sternel Seite gesehen... kannst da ja nochmal hinschauen ;)

Geb mal Rückmeldung obs geht.

Und ja, ich muss mal Bibliotheksversion erhöhen ;)

Grüße Stefan

Tommilein, 2019-07-11:
Danke für die schnelle Antwort,
Was ist die "Sternelseite"? Ich habe mal oben auf der Github-Seite (wo auch noch watch und fork stehen) Star angeclickt. Meinst Du das? Würde gerne noch 10 mal clicken, aber das toggelt wohl nur...

Im Bibliotheksverwalter sieht es bei mir auch so aus wie bei Dir.
[image]

Allerdings wird ja **_"cmperrors interface"_** angemeckert. Im Bibliotheksverwalter sind wirklich nur alte Versionen festzustellen. Ich wüßte allerdings nicht, wie ich diese Bibliotheken einzeln updaten könnte, falls sie tatsächlich veraltet sind.

[image]

Wenn Du eine neuere Version von **_"cmperrors interface"_** bei Dir findest, sag bitte mal Bescheid, vielleicht auch mit einem Hinweis, wie man die einzelne Bibliothek findet zum updaten.

Viele Grüße
Thomas

stefandreyer, 2019-07-11:
Hallo Thomas,

ja, das war der Richtige Stern, Danke!

Ich glaube es ist ein Problem, das ich in dem Testprojekt eine 64 Bit Windows PLC CPU habe und du nur ein 32 Bit Codesys und damit eine 32 Bit Windows PLC CPU.

Mach mal einen rechts Klick auf HomeContrl(Codesys Control Win V3 x64)
Dann Gerät aktualisieren,
Dann Doppelklick auf CODESYS Control Win V3 , ohne X64!!!

Dann hast du ne 32Bit CPU, Übersetzten und testen.

Grüße Stefan

Tommilein, 2019-07-11:
Böse Falle ...
Bei Nutzung des Gerätes CODESYS Control Win V3 (ohne X64) kommen 3 Fehler:
[image]

Beim Raspberry geschieht das Gleiche !!!

[image]

Kann es sein, das LREAL_TO...  nicht mit 32 Bit geht ??  Da hätte ich ein echtes Problem !!

Denn für CODESYS Control Win V3 X64  benötige ich das Ganze nicht.

Da bin ich überfragt.

Noch eine böse Falle:  neues Projekt für Raspberry angelegt und programme rüberkopiert. Nun kommen zig Fehler, daß kein "dynamischer Speicher" definiert ist. Ehe ich das raus hatte ...

Bin jetzt ratlos wegen LREAL.....
Grüße
Thomas

stefandreyer, 2019-07-11:
Hmmm,

ich schau mal, aber erstmal soviel dazu,

auf meinem Raspi läufts, und der ist auch 32 bit.

Dürfte eigentlich kein Thema sein wenn target und Entwicklungsumgebung unterschiedlich sind, oder gleich.

Grüße

stefandreyer, 2019-07-11:
Lief bei mir soweit:

[image]

stefandreyer, 2019-07-11:
Hast du die Oscat Basic lib aus meinem Profil?

Grüße

Tommilein, 2019-07-11:
..._Hast du die Oscat Basic lib aus meinem Profil?..._

Das ist die Frage, wie ich Deine "scharf" schalte. Ich habe ja mehrere solche " Oscat Basic lib"s im Bibliotheksverwalter. (Alleine schon von vor der zeit von MQTT).

Tommilein, 2019-07-11:
Das meine ich: die Bibliothek ist zweimal enthalten.  Wenn ich die obere "selbständige" lösche, kommen sofort Syntaxfehler.
[image]

Mit solchen Bibliotheksfeinheiten bin ich nicht vertraut - eher erledigt...

Tommilein, 2019-07-11:
Heureka,
wenn ich die obere BASIC-Lib rausnehme und nochmal DEINE ins Repository bringen und dann hinzufüge, läuft der "Codesys Control Win 3"  durch:  0 Fehler !!
Und das, obwohl alles die gleiche Version ausweist - ziemlich riskant.

Den Raspberry muss ich morgen weiterüben - der wird wohl dann auch gehen... (ich habe aus Versehen das Programm beendet und nicht gesichert. Das ist eben die Arbeitsdauer ...)

Vielen Dank für Deine Hilfe. 
Morgen fasse ich mal meine Probleme zusammen. Das ist nämlich erst dann ein echter Test, wenn es jemand anderes testet als der Entwickler - der weiß ja immer schon alles. Ich muss mir erst alles tapfer erkämpfen. Aber zuwas macht man sowas sonst ?

Viele Grüße
Thomas

Tommilein, 2019-07-12:
Hallo Stefan,
das gröbste ist geschafft, Dein Beispielprogramm läuft und sendet Telegramme in die Welt (nachgewiesen über MQTT.fx).
Auch empfangen tuts. 

Das das recht holprig ging, liegt wahrscheinlich an meinen geringen Codesys-Kenntnissen. Aber manches ist auch recht kompliziert:
Beispiele: 

"Mal eine neues Projekt mit dem Raspberry machen..." .  Das geht schon , aber... man muss wissen wie man dynamischen Speicher anlegt.

Ich weiss immer noch nicht, wo der Anfang der Topics definiert ist, also das "Device/Application/..." . Ich möchte andere Pfade nutzen. 

Das Problem mit einer erweiterten "BASIC"-Lib (Oscat) ist nicht schön, wenn man im Codesys selbst nicht feststellen kann, ob man nun die neuere Bibliothek geladen hat oder nicht. Nach vielem Suchen kann ich anhand des Änderungsdatums in etwa mitbekommen, welche Lib ich nun habe, welche nicht.
Die Oscat-Lib ist von Dir erweitert worden ??

Dasselbe trifft auf Deine MQTT-Lib zu. Da sie die gleiche Versionsnummer usw. hat wie die vorherige, muss ich nun solange die Libs bei den verschiedenen Projekten rein und rausnehmen, bis ich "nach" bin mit dem Einbauen der neuen MQTT-Lib. Wie schön wäre es gewesen, wenn es eine Abwärtskompatibilität gäbe - einfach die neue Lib rein und die alten Projekte wären gelaufen - für Dich bestimmt kein Problem. Es gibt schon Nutzer, die Deine Lib nicht nur testen , sondern auch in größeren Projekten nutzen !!!

Du hast Dir sehr viel Mühe gegeben, das Beispiel nahe an die aktuellen Geräte zu bringen (ich nehme mal an, sowas wie Sonoff o.ä.). Ich will aber einfach einen String übertragen - wie geht das denn einfach ?
Mit "Methoden" habe ich noch nie gearbeitet, aber mein Hinterkopf sagt mir, das genau in der Methode "PublishRecive" das Text-Übertragen verborgen ist.

Eine kleine Visualisierung hätte das testen für solche Geister wie mich alles etwas anschaulicher gemacht. Ich hatte zu tun, irgendwie eine Variable zum Togglen zu finden, die die Lampe enfdlich mal geschaltet hat. Vorteil war, ich musste mich intensiv mit den Programmen beschäftigen - da lernt man kräftig!!.

[image]

Jetzt will ich aber mal aufhören, mich zu beschweren. Das sollen ja auch nur Hinweise sein, was mir aufgefallen ist, jemandem, der nun nicht zu den Codesys-Gurus zählt. 
Soweit erst mal bis hierher. Vielen Dank dafür, das Du so eine Lib überhaupt ins Netz stellst. 

Viele Grüße
Thomas

stefandreyer, 2019-07-13:
Hallo Thomas,
schön das es bei vorrangig geht.
Zum Thema dynamischen Speicher: das steht auf der Start Seite der Bibliothek und es gab auch schon ein issue dazu. Ich kann mir vorstellen das der einstieg so aus der kalten ins codesys und diese Bibliothek nicht einfach ist. Aber ich kann und will hier nicht noch ein codesys Tutorial platzieren. Das machen schon andere prima. Gerne aber beantwortet ich Fragen im Kontext der Bibliothek.
Im GreatExampleOfAdvanteges wird das Topic aus den Namen der Applikation und FB Instanzen generiert. Bringt den Vorteil das man auch  Instanzen einfach in ein anders Topic verschieben kann oder einfach umbenennt und die folgenden einfach mit.
Einfach einen string senden machst du mit dem FB MQTT publish(oder so änlich, bin nicht am pc). Mit Methoden wirst du dich in dem Zusammenhang auch beschäftigen müssen. Gehört zu codesys V3 halt dazu. Sind nur Funktionen einer "Klasse".
Ich glaub ich habe nix an der Oscar lib gemacht. Nur den letzten stand von v2.3 nach V3 portiert. Warum die jetzt unterschiedlich sind ???? Aber ich lobe Besserung bei meinen Bibliotheks Versionen. Github zusammen mit codesys macht es einem da nur nicht so einfach. Glaub ich ;)
Und, klingt so als wenn du eine visu für mein Beispiel bauen willst ;) besser kann man nicht lernen...
Viel Spaß noch und lass ruhig mal hören wies bei dir klappt.
Grüße Stefan

Tommilein, 2019-07-13:
Hallo Stefan,
vielen Dank für die Antwort. 
Wenn Du wieder mal am PC bist, schau mal, was ich alles vor einem "publish" machen muss. Da gibt es (leider für mich) eine Menge Abhängigkeiten. Topic und Payload kann man ja direkt vorgeben, aber innerhalb der Lib benötige ich noch Daten wie MQTT_IN_OUT .BROKER_CONNECTED  werden irgendwie "aussenrum" übergeben. Aber ohne dieses Flag geht nichts los...
Wie wird MQTT_IN_OUT  vernünftig übergeben?

Vielen Dank
Thomas

stefandreyer, 2019-07-13:
Hallo Thomas,

schau mal in die anderen Github Beispiele, die sollten recht schmal sein.

Grüße Stefan

Tommilein, 2019-07-14:
Hallo Stefan,
hast recht, die anderen Beispiele sind viel schneller zu begreifen  -   aber !!

In Deiner alten Bibliothek gab es ein Flag "subscribe.Recived" zum Zeiche, daß etwas empfangen wurde. Dieses Flag gibt es nun nicht mehr und ich probierte es mit "subscribe.SubscribeDone" aus. Ich glaube aber nicht mehr, das dies das richtige "Empfangsflag" ist - es bewegt sich nicht:

[image]

Mit welchem Signal kann ich denn nun bestimmen, das etwas subscribed wurde ??

Vielen Dank
Thomas

stefandreyer, 2019-07-14:
Hallo Thomas,

hmmm, ich interpretiere mal. Eigentlich willst du wissen ob was auf die Subscription empfangen wurden. Das bit SubscriptionDone singaliesiert, das die angeforderte subscription vom broker akzeptiert und eingerichtet wurde.

Was du wissen willst ist der empfang eines publish vom Broker. Diese Konzept habe ich aus diversen Gründen umgestellt. Wenn nun etwas passendes zur subscription empfangen wurde wird der am Eingang Callback hinterlegte FB aufgerufen. Das nennt sich callback. Diese FB kann nun eine einzelne Variable auswerten und beschrieben oder eine ganze sammlung(collection). Im konkreten Beispiel kümmert sich der FB StringCallback darum und schreibt die empfangenen Daten an die gewünschte Position.

Wenn du noch Fragen hast, frag ruhig.

Grüße Stefan

Tommilein, 2019-07-15:
Hallo Stefan,
erstmal vielen Dank, daß Du selbst zu nachschlafender Zeit mir noch antwortest. Ich find's toll.

Ich habe jetzt ein Programm so umgestellt, das ich Texte zum Broker hin schicken kann und sie wieder lesen kann. 
Ich habe dabei die größten Probleme. Deine Programmiermethoden lassen nicht zu, das ich einzelne Teile "herauskopiere" und zu einem einfacheren Programm zusammensetze. Du verwendest innerhalb Deiner Funktionen sowas wie globale Variable (nicht echte, aber irgendwie übergeordnete Variable).
Hier ein Auszug aus einem abgewandleten **"LampBase":

VAR
        //für extra publish
	publish:SD_MQTT.MQTTPublish;
	QoS:SD_MQTT.QoS;
	send: BOOL:=False;;
	SendDone:BOOL;
	RetainMqtt: BOOL;
end_var

...
...

IF SwitchTrigger.Q THEN
	 send:=TRUE;
END_IF

// SendState();     // SendState() will ich ablösen, um einfach los senden zu können
                          // und nicht via getoggeltem Status was rauszusenden.

// ab hier jetzt eigenes publish  -- und das funktioniert nun nicht...
publish(
	Topic:= ADR(MyTopicSend), 
	PayloadString:= ADR(TextSend), 
	PublishAsString:= TRUE, 
	QoSIn:=SD_MQTT.QoS.ExactlyOnce,
	MRetain:= RetainMqtt, 
	send:= Send, 
	done=> SendDone);
	
IF SendDone THEN
	send:=FALSE;
END_IF
...
...

Hier liegt irgendwas in der Umgebung von "publish"   NICHT vor. Jedenfalls bewegt sich so nichts, obwohl das mehr oder weniger äquivalent zum Inhalt von SendState() ist.

Viel schlimmer ist aber noch, das ich keine 2 Clients einrichten kann. Mit Deiner alten Lib.-Version habe ich einfach 2x MQTT instanziert.
Hier bezweifle ich, das ich sowas hinbringe, da  doch wahrscheinlich das hier dem entgegensteht:
VAR_GLOBAL
	MQTT_IN_OUT:SD_MQTT.MQTT_IN_OUT;
END_VAR

Wie sollen hier die beiden Clients unterschieden werden. Ich glaube auch, das alles Funktionen sich auf EINEN Client beziehen, da ja nichts übergeben wird. Jedenfalls verstehe ich es nicht besser.

Um es zusammenzufassen:

Ich bauche Hinweise für eine richtige Umgebung für die Lib-Aufrufe 
und  wie ich 2 Clients erzeuge und anspreche.

Viele Grüße
Thomas

PS:
Ich stamme noch aus einer Zeit, da man Wert darauf legte, das eine Funtion nur dann"sauber" programmiert war, wenn sie unabhängig "von aussen" funktionierte - also nix global... Das war aus heutiger Sicht eben schön altmodisch - nix objektorientiert.
Alle Parameter wurden mitgegeben und alle Ergenisse standen als Funktionwert oder Ausgabevariable zur Verfügung. Sowas ließ sich noch debuggen und wenn man die Quelle hatte, auch leicht lesen.

Jetzt stelle Dir mal vor, was das für eine Arbeit für einen wie mich ist, wenn nun Funktionen  gar nicht mehr erkenntlich aufgerufen werden und trotzdem dann online benutzt werden (PublishRecieve). 

Zum Glück hatte ich auf der Arduino-Strecke mal die Callback-Funktion im Fokus. Aber da gab es einen Parameter mit einem Aufruf einer Funktion, die 5 Zeilen weiter unten auch zu finden war.

Bloß das Du verstehst, warum ich hier so rumeiere...

stefandreyer, 2019-07-15:
Hallo Thomas,

es ist wieder abends...

Scheint so als wenn du gerade voll in die OOP mit meiner Bibliothek reinschlitterst ;) Du wirst noch viel Freunde haben.

Ich kenne auch noch eine Zeit mit SPS Programmen in STEP 5 von 1989. Und die waren sauber programmiert, haben dich aber dank indirekter Programmierung zum Wahnsinn getrieben... Soll heißen: jede Programiersprache kann man versauen, aber einige Ansätze sind sehr hilfreich und unerlässlich. Und mitlerweile kann ich CODESYS ganz gut debuggen, dir fehlt hier warscheinlich noch das Handling, aber das wird noch.

Ich habe auch eigentlich alles gut gekapselt und versucht in FB's zu verstauen. Globaler Kontext ist da sehr gering. Sicherlich vorhanden, aber für Fixwerte hauptsächlich.

Das Variablen aus dem nichts beschrieben werden war auch vorher schon möglich und auch sinvoll.

Die von dir angesprochene "Globale variable" ist warscheinlich ein Objekt das ich angelegt habe, damit alle Instanzen mit der Instanz des Cliente Sprechen können. Das geht auch in OOP nicht besser, Irgendein immer verfügbares objekt brauchst du immer, und sei es das du es dir mit einer get Methode holst.

Bei deinem publish kann es sein das du die Init Methode noch nicht aufgerufen hast. Da bekommt die Instanz einen Zeiger auf die globale variable um mit dem Client zu reden. Und ohne diesen Init macht publish einfach nix... Da werde ich auch nochmal ansetzten, das die Intialisierung beri der deklaration mitgegeben wird, da kann man das dann nicht vergessen.

Zum Thema 2 clienten Ich habe eigentlich nix Programmiert was dem entgegen steht. Ich weiß jetzt nicht wie genau du deine 2 Clienten und Variablen bei der Bibliothek versorgt hast aber mit der aktuellen ist das auch möglich.
Das:

VAR_GLOBAL
MQTT_IN_OUT:SD_MQTT.MQTT_IN_OUT;
END_VAR

schränkt dich da nicht ein. Das kann dazu werden:

VAR_GLOBAL
MQTT_IN_OUT_CLIENT1:SD_MQTT.MQTT_IN_OUT;
MQTT_IN_OUT_CLIENT2:SD_MQTT.MQTT_IN_OUT;
END_V

Wie du das dann danach verteilt hast würde mich auch interessieren.

Hau rein ;)

Grüße Stefan

Tommilein, 2019-07-16:
Hallo Stefan,
ich bin wieder weiter:  3 Clients funktionieren. 
[image]

Ich habe alles 3x angelegt (GVL)
Aber nur eine MQTT_STUFF , da drin liegt auch alles Erforderliche drei mal.
So funktioniert das dann:

[image]

Ich habe aber kein gutes Gefühl, weil noch sowas auftritt:

[image]

ich habe hier ganz schnell "Senden" geklickert.  In dem Moment geht ein Empfangstelegramm verloren - die Nr. 12. Da gehen auch manchmal noch mehr verloren. Es aknn aber auch an mir liegen.

Wahrscheinlich liegt es am Senden, denn am Broker kommen auch nicht alle Telegramme an.

Ich habe auch noch das "publish" - Problem. Es gibt ja wohl kein MQTTPublish.Init  . Wo oder wie muss denn dann das MQTTPublish  initialisiert werden.
Aus der originalen Funktion sendstate()  kann man keine besonderen Initialisierungen erkennen (ich wenigstens nicht). Das "SendState.Init"  ist in jedem Falle drin. 

Kann Du mir nochmal einen Hinweis diesbezüglich geben ??

Vielen Dank für deine Mitwirkung

Viele Grüße
Thomas

stefandreyer, 2019-07-16:
Hallo Thomas,

schaut ja schon gut aus.

Wir bekommen noch die Kuh vom Eis und das Kind noch geschauckelt.

Die Init Methode die ich meinte heißt ` SetMqttInOut`.

Wie hast du die Broker getrennt? Ich würde dir pro Broker  einen Client FB und eine MQTT_IN_OUT  Struktur sowie pro Variable , einen publish FB, eine  und einen subscription FB empfehlen. Unter dem Subscription FB können natürlich mehrere Topics bearbeitet werden.

Den callback FB kannst du jedoch für alle Broker nutzen. Das passt soweit.

Wie hast du das aktuell organsisiert? Vielleich bekomme ich so raus, warum da was untergeht.

Wenn du sendState nicht mehr brauchst kannst du die instance und alle Aufrufe dazu wegwerfen.

Das du alles in MQTT_STUFF packst passt schon.

Und schön das du die verschiedene broker  unterstrichen und dann verpixelt hast ;)

Grüße Stefan

[8 more replies]

closed 2019-09-23 by stefandreyer