stefandreyer/CODESYS-MQTT #9: Question about compatibility with SoMachine 4.3

Javierete, 2018-11-05:
Hello,

I want to add your MQTT library to SoMachine 4.3 but thows dependencies errors (missing libraries):
[image]

As I have Codesys development too, I've tried to add the libraries from Codesys but I suspect that SoMachine uses an older version of Codesys.
Do you have experience with SoMachine? Will try to make it work, but I have little experience dealing with libraries...

stefandreyer, 2018-11-05:
Hi Javierete,
the only lib I can tell you is VommonTypsAndFunctions, this is available in my GitHub Account. The other libs, phu, I have no experiences with SoMachine, till now I didn't know that this is Schneider PLC Software. Maybe you are able to set the library Version to the Version offerd you by Schneider. 
Next, are all needey librarys offerd by Schneider?

BR Stefan

closed 2019-02-21 by stefandreyer

Albalate, 2019-10-27:
hola JavierNR.
lograste hacer funcionar en SoMachine la librería?