stefandreyer/CODESYS-MQTT #11: use username and password

zhapwey, 2019-02-20:
Hello:
First of all congratulations for the work
I have tested your library for mqtt and it works very well. I tried to mount some tests in cloudMQTT that requires username and password and I have not managed to publish or subscribe to any topic.
Could you tell me how to do it?

stefandreyer, 2019-02-20:
Hello Zzhapwey,

i have to admit that i implemented username and password, but never tested it...
so first look if the tls Connection is established. So have a look into the Ethernet by wireshart, or look into MQTT_CONTROLL if step is at least 30 TLS Connection is done.

BR Stefan

SilviuMarchidanu, 2019-04-02:
Hi Stefan , 
How can we set the user and the password ?
I can see it in the COM .
[image]

stefandreyer, 2019-04-02:
Hi Silviu,

i need to correct here something ....

BR Stefan

stefandreyer, 2019-04-02:
Hi Silviu,

solved. Please use this for username and password in HANDLE_MQTT:
	`//use format: user:password@domain:port`
	`URL : STRING(OSCAT_BASIC.STRING_LENGTH);`

BR Stefan

SilviuMarchidanu, 2019-04-03:
Hi Stefan , 
Thanks for looking in to this .
You have made already an amazing job by creating this MQTT FB .
I will give it a try tonight and report back .

closed 2019-06-29 by stefandreyer