stefandreyer/CODESYS-MQTT #13: SONOFF TASMOTA

SilviuMarchidanu, 2019-03-29:
Hi Stefan, 
Whats the best way to have multiple instances of the  subscribe function .
Have been trying with  the example "TestMQTTGithubInterfaceExampleTopicAndPayloadRaspberry" but for the SONOFF with the Tasmota Firmware the topic are quite long.
For example I need to extract the payload from the the fallowing topics  
HOME/GARAGE/tele/STATE { "Vcc":3.12, "Power":ON, "Level":High }
HOME/GARAGE/stat/POWER ON

HOME/IRIGATION/tele/STATE { "Vcc":3.12, "Power":OFF, "Temperature":21.7 }
HOME/IRIGATION/stat/POWER OFF

Any help is much appreciated.

stefandreyer, 2019-03-29:
Hi Silviu,

please Take a look into the GreatExampleOfAdvantages.project. I Think i Would make some SONOFF FBs, build in some nice topic creation(Cut out Project and Application) and built my Topic tree by instance naming and program struture. In this FBs you can split the Payload for your values in the publishRecived method.

it's meaningful too, to insert a "recive" and "send" Topic Level below "Home" like:

HOME/SEND/GARAGE/tele/STATE { "Vcc":3.12, "Power":ON, "Level":High } (look by SONOFF)
HOME/RECIVE/GARAGE/stat/POWER ON (look by SONOFF) (is this a command?)

else your subscription would recive anything you have published

Then i wolud subscibe to Home/Send/# and the client is sorting the topics. 

can you post a full topic? 

Have Fun!

BR Stefan

SilviuMarchidanu, 2019-03-29:
Thanks for your quick reply Stefan, 
The structure of the SONOFF TASMOTA Mqtt Topics are the fallowing by default , but can be moved around .
Each Sonoff device has a unique name which you configure in the topic section, this name will reflect the use of the device and should be meaningful e.g : Garage_LIGHT, IRRIGATION , FRONT_DOOR_LIGHT etc.
The prefix is either cmnd, stat or tele  and the choice is usually topic+prefix (%topic%/%prefix%/). or prefix +topic (%prefix%/%topic%/) .
Most of the times I add the topic for the area where they are installed as well so it will come out as : area+topic+prefix .
examples:
HOUSE/tele/Garage_LIGHT/ or HOUSE/Garage_LIGHT/tele

Using the prefix as the first part of the topic makes it easier to subscribe to multiple sonoff devices using:
tele/#, stat/# and cmnd/# options.#

So if all the Sonoffs were used in your house you could use the topic hierarchy
house/prefix/topic or house/topic/prefix and can subscribe to all Sonoff devices using house/#
In this case I prefer house/topic/prefix.

MQTT Default Topic Prefixes : 
cmnd = PC,Smart Phone using an MQTT client to control the Sonoff,set configuration and get  status information
stat = Sent by the Sonoff , Sent in response to a command
tele = Status information , Report status information at set intervals

Some real examples:
 In response to TOPIC = HOUSE/Garage_LIGHT/cmnd/POWER1  , Payload : on
Subscriber gets on TOPIC = HOUSE/Garage_LIGHT/stat/POWER1 , Payload: ON

Or the tele , that is set at intervals in Sonoff in a JSON format:
Subscriber gets as per the tele interval TOPIC =  HOUSE/Garage_LIGHT/tele/STATE , 
Payload: { "Time":"2019-03-29T18:30:20","Uptime":"1T01:06:31" , "Vcc":"3.170", "POWER1":"ON", "WFi":{"AP":1,"SSId":Home_Net","RSSI":90, "APMAc":"FF:FF:FF:FF:FF:FF"}}

On the tele Payload in the Sonoff config you can add more variables that can be transmitted as per the Sonoff type or sensors attached like : 
"Temperature":21.7 or "Humidity":70 or "POWER2":"OFF" 

Looking Forward in to this matter.
The GreatExampleOfAdvantages.project has been created with the latest SP 14:10 , need to update my Codesys :))

SilviuMarchidanu, 2019-03-29:
Hi Stefan, 
Have installed Codesys SP14.10 ans still not able to open the GreatExampleOfAdvantages.project any chances that it might be corupted ? Maybe some other tips of your setup ?

stefandreyer, 2019-03-30:
Hi Silviu,

no Problems here, what is your error?

BR Stefan

SilviuMarchidanu, 2019-03-30:
[image]
Hi Stefan , Please see attached.

stefandreyer, 2019-03-31:
Hi Silviu,

here is this version.

BR Stefan

[image]

stefandreyer, 2019-03-31:
Hi Silviu,

i think you can build your topics like the example and then you can subscribe to:

HOUSE/+/cmnd/#

If you need more tips, then feel free to ask.

I think sonoff is nice for my home too....

BR Stefan

SilviuMarchidanu, 2019-03-31:
Hi Stefan , 
I had difficulties understanding the MQTT Subscribe and Publish but now is more  clear   and started to understand it after checking some of the examples.
I'm in the process of building some FB to integrate your MQTT Binding with the SONOFF TASMOTA MQTT Format. I'm not an expert but when I have something I can have it sent to you to check it out .
Any chance of sending me the GreatExampleOfAdvantages.project saved in a Sp13 Format ?

stefandreyer, 2019-04-02:
Hi Silviu.

SP 13, i'll have a look.

BR Stefan

stefandreyer, 2019-04-08:
Hi,

I made a test file with 3.5 sp 11. Please test it.

BR Stefan

SilviuMarchidanu, 2019-04-08:
Cheers Stefan , Ill give it a try .
I've been Busy with the sonoff stuf . 
I have made some FB for the Sonoff Basic , 4 CH and TH . 
I will upload it in a few days on my account.

closed 2019-11-26 by stefandreyer