stefandreyer/CODESYS-MQTT #14: Undissolved references with using EXC88 (Eckelmann)

SvenJoedicke, 2019-05-02:
Hallo,

we use your MQTT Library successfully with Raspberry PI and we are satisfied with it. But the same source code creates reference errors when the plc goes to run. 

wir nutzen die MQTT-Bibliothek sehr erfolgreich und mit Begeisterung mit dem Raspberry PI. Nun müssen wir aber diese auch auf unserer hausinternen Steuerung von Eckelmann zum laufen bekommen, bekommen aber sobald die Steuerung in den Run geht, Ausnahmefehler bezüglich unaufgelöster Referenzen. Mit Eckelmann konnten wir schon diverse Sachen eingrenzen, was Richtung openSSL betrifft. Jedoch gibt es immernoch diverse Probleme. Ist es möglich sie diesbezüglich offiziell zu kontaktieren, zum Beispiel über XING??? 

Ein Support wäre echt cool und ich würde es mir für diese tolle Lib echt wünschen, dass sie weiter getragen wird. Wir sehen in dieser großes Potential.

stefandreyer, 2019-05-02:
Hallo,

haben sie meine Mail erhalten?

Vielleicht im Spam?

Grüße

SvenJoedicke, 2019-05-02:
Hallo,

Also auf die Email von 11 Uhr habe ich eigentlich geantwortet.

MFG
Sven Jödicke

sendet von Mail für Windows 10

Von: stefandreyer
Gesendet: Donnerstag, 2. Mai 2019 17:41
An: stefandreyer/CODESYS-MQTT
Cc: SvenJoedicke; Author
Betreff: Re: [stefandreyer/CODESYS-MQTT] Undissolved references with usingEXC88 (Eckelmann) (#14)

Hallo,
haben sie meine Mail erhalten?
Vielleicht im Spam?
Grüße

closed 2019-07-01 by stefandreyer