stefandreyer/CODESYS-MQTT #15: Subscribe to MQTT Topic

PHSx76, 2019-05-21:
Hi Stefan
Thank you very much for your works!!!
I´m new with MQTT so sorry if is a dumb question.
I´m able to connect to the broker (mosquitto), I have the simulation 
mosquitto_pub -h 127.0.0.1 -t mytopic -m MessagetoCodesys
From Codesys I can subscribe to a Topic mytopic.
But, I don´t see the publish message, what is the right function to read the message
I have tried to adapt the demo project, but unable to read the publish message where I´m subscrite, from codesys only want to read messages, not public messages.

Thank you very much for your time
Best Regards

stefandreyer, 2019-05-21:
Hi PHSx76,

I see I have to update the wiki… ;)

OK, you found the subscription FB. This is fine. This FB makes the init of the subscription packet to the broker. Then, if the codesys Client recives a publish packet, the Client checks if the publish Topic Matches the subscription topic. If so, the Client calls the callback FB given to subscription FB(Input Callback at subscription FB).

This callback FB Needs to implement the MQTT_SUBSCRIBE_CALLBACK interface.

The main Advantage of this interface is the callback_collector. To this collector multiple instances can be added. So Each FB can add itself. You can check this out in the greatExampleOfAdvanages.

For easy use of the callbacks I made some FBs for strings and values: ReciveString; ReciveValue; ListStringCallback; ListAllValuesCallback.

You can Inits this FBs with the Topic or payload whicht they have to chek the match of the topic. If the recived publish matches to the match param then the value is beeing updatet.

And this is the power of the library, the matching Topics can be createt by the structur and instance names of the programm. Now see again the greatExampleOfAdvantages. There the subscription is made to 'HomeControl/Application/Recive/#'. Then eache FB instanc checks if the topic matches.

Sounds difficult? It may looks so, but at the end it makes it more easy....

Got more questions? Feel free…

BR Stefan

PHSx76, 2019-05-22:
Thank you very much for your reply Stefan :-)

Yes, I have more questions... :-)

I use one off your program, like you said (ReciveString; ReciveValue) because still I don´t know where I receive the message, please see the attach pictures.
I´m connected to the broker (mosquitto) and I´m subscripted to one topic. But when I publish a new data I don´t see this message in CodeSys
If I understand, I must see the data here "MQTTString.MQTTRecived" it is right?
But I don´t know what I have doing wrong.
With the function states() I can publish data with the clock signal, but as I said, I don´t know why I can´t see the new messages . 
Did you see anything wrong in this screenshots? 
Thank you very much for your time.
Best Regards
[image]
[image]

stefandreyer, 2019-05-23:
Hi Hi PHSx76,

yes, you should see the the data at the Point you marked. What I see is that you don't have initialisized the reciver. You have to call initAsFindPayload or initAsFindTopic to init the reciver for what he have to look. 

The best way to check what the Client is reciving, make a brakepoint at first line of the PublishRecived method of the Collecteor and subscribe to '#'. The stept throught the program and check what it is doing....

Have fun!

BR Stefan

PHSx76, 2019-05-23:
Hi Stefan
Thanks again for the tip.
Yes, I called initAsFindPayload.
Still don´t work, but I was able to see something... 
If I put a breakpoint to the PublishRecived method, I can see the data, but nothing in MQTTRecived, see the screenshots. 
By the way attach the project if you have time to see it.
 
[image]
[image]
[MyTestMQTTPayload.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/3212217/MyTestMQTTPayload.zip)

Thank you very much
Best Regards

stefandreyer, 2019-05-23:
Hi PHSx76,

try to call initAsFindTopic. Like you done, you try to find 'SimulacionTemperatura' in the payload, but it resists in the topic.

Looks like JSON data, you can write your own FB an implement the interface, then in the recive method you can parse the data....

And what I have seen, I have to do rework on the recive FBs, they are not so consistent....

Have fun!

BR Stefan

PHSx76, 2019-05-23:
Thanks again.
Yes, I have tried too. I will waiting the news updates.
All my testing was with the purpose of writing an article.
By the way if you want a help to update the wiki, let me know.
Best Regards

stefandreyer, 2019-05-24:
Hi!

Try this.

[MyTestMQTTPayload.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/3214955/MyTestMQTTPayload.zip)

Publish to : WinCC/Training/SimulacionTemperatura
broker: test.mosquitto.org:1883

BR Stefan

stefandreyer, 2019-05-28:
Hi PHSx76,

what Kind of articel do you want to write? Did you have succsess with my attaced projekt?

BR Stefan

P.S. Help for wiki would be fine...

PHSx76, 2019-05-29:
Hi Stefan
Sorry, the same behavior with the last demo, I don´t know if could be because you are running x64 and I have trying with x86.
I don´t comment because I don´t want bother you more with this issue, I don´t know if this only happen to me, so I don´t want bother you.
I write article for my web site https://plc-hmi-scadas.com, It is spanish but easy to translate with google translate and I was testing your library because WinCC has this new feature with the last version v7.5 https://plc-hmi-scadas.com/WinCC_MQTT.php.
If you want, you will see my mail in the website and you can contact me, and we can talk.

Thank you for your time.
Best Regards

stefandreyer, 2019-07-02:
Hi PHSx76,

did you try again? x64 and x86 shouldn't be a problem.

Still Need help?

BR Stefan

closed 2019-11-26 by stefandreyer