stefandreyer/CODESYS-MQTT #17: question size message

zhapwey, 2019-07-01:
Hello:
Incredible work in the library, its functionality in the number of shipments is simply perfect.
Now I find a problem I have very large messages and I was wondering if it was possible to easily read messages of more than 255 characters.

And this would be awesome for the library!

stefandreyer, 2019-07-01:
Hello zhapwey,

this should allready work fine. The library has no limitations for string sizes. The Pointer to the payload string may be declared to 255 Bytes but can hold more than this. The string is allways terminated by a zero. 

You may have problemes watching this strings in debug windows, but it is valid. Use memory windows instead. The codesys string libs only supporting strings < 255 Byte too. So I'd wrote some string functions in CommonTypsAndFunctions for strings bigger than 255 Bytes, like find.

Actually the lib is config for max payload size of 2000 bytes. This is your max string size.

You are able to change this size in:
`TYPE COM_CONFIG :
STRUCT
	//max payload size
	MaxPayloadSize : UDINT := 2000;
	//max topic levels
	MaxTopicLevels:UINT := 25;
	//Max Topic elemt size
	MaxTopicElementSize:UINT:=255;
	
END_STRUCT
END_TYPE`

This resists in your MQTT_IN_OUT var.

Try it, report me and give me a star ;)

BR

Stefan

zhapwey, 2019-07-02:
hello:

Yesterday's tests change the definitions of several strings in the string library (255) one string (2048).

MQTTSubscribe
PayloadString: STRING (2048)

TestOutOfTheBoxTLS

the message where you save in subscribe payload.

Now that I have updated the library with callbacks I am going to try it because I was using an older version.

regards

I put your well-deserved star.

zhapwey, 2019-07-02:
hello:

We understand your message in the place. To work with large strings that string libraries do not work with more than 255 characters.
I am now working with the stringutils that allows me to work with string pointers and the limitations are much lower,

Thanks for your help,

stefandreyer, 2019-07-02:
Hi,

Sounds great, please let me know if it works as expected, or not ;)

BR Stefan

closed 2019-09-23 by stefandreyer