stefandreyer/CODESYS-MQTT #18: No memory Error when trying notls on wago pfc100

sanderobdeijn, 2019-07-06:
Has anyone tested this on the wago pfc 100?
It tried to execute TestOutOfTheBoxNoTLS and on build it says.
C0265:  No memory for dynamic object creation defined for application 'Device.Application'

Does anyone know what this means and how to fix it?

stefandreyer, 2019-07-06:
Hi Sander,

Right click on application propertis there somewehre you can set the dynamic memory.

My projekt uses dynamic allocated memory, but you have to activate it.

Did you use the Projekt or the FB?

BR Stefan

sanderobdeijn, 2019-07-06:
Thank you that was the problem.

sanderobdeijn, 2019-07-06:
btw stefan what do you mean with Projekt or the FB.
What is a FB exactly

stefandreyer, 2019-07-06:
Hi Sander,

Project i mean the test Projects you can find in my repository. FB are the POUs which called in the projects.
FB --> German: Funktionsbaustein. englich fucntionblock???

BR Stefan

stefandreyer, 2019-07-06:
Hi,

please test if test.mosquitto.org is online...

often it is not ...

then use somthing like mqtt fx to check the messages on the broker.

In MQTT_IN_OUT you'll se if the client is connected to the broker.

there is a var BROKER_CONNECTED if true, client is connected.

BR Stefan

closed 2019-07-08 by stefandreyer