stefandreyer/CODESYS-MQTT #19: Error VAR_IN_OUT 'MQTT_INFO' must be assigned in call of 'HANDLE_MQTT'

sanderobdeijn, 2019-07-06:
I'm looking at the GreatExampleOfAdvantages project
In the MQTT_STUFF (FB) is this piece of code.

Client( 
	MQTT_IN_OUT:= GVL_MQTT.MQTT_IN_OUT, 
	ENABLE:= TRUE, 
	URL:= 'test.mosquitto.org:1883', 
	TIMEOUT:= T#10S);

When I build the project I get the following error.
C0039:  VAR_IN_OUT 'MQTT_INFO' must be assigned in call of 'HANDLE_MQTT'
I think the MQTT_INFO must be set in the function call?

stefandreyer, 2019-07-06:
hmmm,

Looks like I didn't update the example to the new lib version, I'll check this…

BR Stefan

stefandreyer, 2019-07-06:
Hi,

I Updateted the projekt, Looks like I didn't made the upload. And test.mosquitto.org is acually dead....

Let me know if its looks fine for you...

BR Stefan

closed 2019-09-23 by stefandreyer