stefandreyer/CODESYS-MQTT #25: hundreds of errors when built

PLCMercenary, 2019-10-06:
tried using all of the examples and tests, tried to execute every one of the "github" tests as well. Hundreds of errors all showing ambiguous use of "find" and ambiguous use of "TON"

I do MQTT and PLC communications for a living. And I have never been so frustrated and struggled so hard to publish some topics then trying to use Codesys and any one of 4 or 5 different libraries including the demo from 3S. 

Please Mr Stefan, point me in the right direction. I am totally lost at using libraries on windows 10 or Windows 7 64 bit with Codesys 3.5.15 
[image]

PLCMercenary, 2019-10-06:
[image]

stefandreyer, 2019-10-07:
Hi,

please open the library and check the libs.

I use Version 3.5 SP 14. You sholud have success when you update to your version.
All blue warings you can ignore.

For Easy use of the Liebe have a look to:

https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3

If your libs are clean it sholde be easy to publish your stuff.

BR Stefan

PLCMercenary, 2019-10-20:
Its not your lib.. the issue is with the Crypto handling and the packages Opto22 built into their run time and their library.. Your Library work perfectly when I run it off of the Wim V3 on my desktop. and the only MQTT that will work on the Opto is the Janztech, because they have 2 seperate libs for TLS and Non TLS. 

Have you considered or do know of at Sparkplug package forCodesys??

stefandreyer, 2019-10-21:
Hello,

I will not provide 2 versions of the lib. 

Your solution:

comment out all apperance of TLS stuff in IP_CONTROL in your local copy of commonTypesAndFunctions.

Deleting the lib CmpTls out of library manager should show you where.

[image]

This should work for you.

BR Stefan

PLCMercenary, 2019-10-29:
Thanks Stefan.. That likely will work.. But I will just wait for Opto22 to fix their Runtime.  I would rather use your library AND TLS. They are working on a bugfix now

stefandreyer, 2019-10-29:
Hi,

Sounds perfect…

BR Stefan

closed 2019-11-07 by stefandreyer