stefandreyer/CODESYS-MQTT #26: Codesys MQTT Add Library Problem

wildanfathur, 2019-10-10:
Hello Stefan,
I am interested in using MQTT on codesys using the library that you created. However, when I try to add the library, an error like this appears:

[image]

what should I do?
because I'm still relatively new using codesys

stefandreyer, 2019-10-10:
Hi, 

please Download the latest Library 1.1.0.0. Then try the button Download Missing Librarys.

BR Stefan

wildanfathur, 2019-10-13:
Hi, thanks for the solution, 
there is any basic tutorial on how to use this MQTT library?

stefandreyer, 2019-10-21:
Hi,

there isn't a tutorial yet. Have a look at the examples, they should help you.

I think 32bit codesys should be fine.

BR Stefan

wildanfathur, 2019-10-30:
Ok, but I've try open the GreatExampleOfAdvantages.project, but I found the problem like this:
[image]
I am using CODESYS V3.5 SP15
is there any suggestion for me ?

closed 2019-11-07 by stefandreyer