stefandreyer/CODESYS-MQTT #27: Codesys not connecting to mqtt server

sanderobdeijn, 2019-10-29:
I have a Wago PFC100 and I have a mqtt server running in my local network without authentication.
I have tried the testMQTT(client:={mqttserver:1883} test function but the plc doesn't seem to connect to the mqtt server.
When I look at MQTTClient > MQTT_Control > sResultIp the value is empty.
When I look at #3 I think this value should be set. If this doesn't works it suggest a dns issue?

stefandreyer, 2019-10-30:
Hi Sander,

does you URL_DATA looks like this?
[image]

Do you realy write testMQTT(client:={mqttserver:1883} ?

should look like: testMQTT(client:='mqttserver:1883')

BR Stefan

sanderobdeijn, 2019-10-30:
what i wrote was: testMQTT(client:='192.186.1.83:1883');
And my url data looks like this.
[image]
And I don't see a connected client to my mqtt server.

stefandreyer, 2019-10-31:
Hi Sander,

with a static IP it have to look like:
[image]

With static IP I don't use DNS, this have to work.

You may look in ErrorHistory in HANDL_MQTT, maybe you have a problem with dynamic memory.

If this, I don't stat conecting or spliting connection string.

BR Stefan

sanderobdeijn, 2019-11-01:
Hi Stefan I have set a static ip in the web based management page.
This doesn't fix it.

I have looked at the error history. And you called it.
[image]
It seems to be a dynamic memory issue. 
What does this mean. Is this a hardware issue or a configuration issue?

primsam, 2019-11-01:
> Hi Stefan I have set a static ip in the web based management page.
> This doesn't fix it.
> 
> I have looked at the error history. And you called it.
> [image]
> It seems to be a dynamic memory issue.
> What does this mean. Is this a hardware issue or a configuration issue?

Hi ,
That ist a configuration issue .
You need to increase the dynamic memory 

Right click on the application
- properties
- Dynamic memory settings
- increase maximum size of memory

sanderobdeijn, 2019-11-02:
How much memory do I need? Do you know?

stefandreyer, 2019-11-02:
Hi Sander,

here you see your usage of dynamic memory in bytes:
[image]

If you use more subscriptions and more instances of callback FBs this will grow.

and yes, it could be usefull to set up some setup instructions ;)

BR Stefan

sanderobdeijn, 2019-11-02:
this usage is in bytes or mb?
100 mb wasn't enough for the GreatExampleOfAdvantages project.
my plc only has 256 mb ram does this also limit the dynamic memory?

primsam, 2019-11-02:
This is displayed in Bytes i think.
At least when i limit Dynamic memory at 20000 Bytes it run without Problems

stefandreyer, 2019-11-02:
Hi,

yes these are Bytes. I testet the GreatExampleOfAdvanteges with 20000Bytes(20kByte).

At most I setup my Raspi with 2000000 Bytes and got enough left.

@sanderobdeijn yes your physical memory limits the dynamic memory. But my lib takes only veeery few of it. I think with 1000000(1MByte) you can make almoste everythink with my lib on your plc...
100MBytes(100000000) are more then enough for this example.

BR Stefan

sanderobdeijn, 2019-11-05:
yeah my bad I forgot about Kbytes. I have used 153600 bytes or 150 kBytes. It has been long since I had to calculate in bytes.

closed 2019-11-05 by sanderobdeijn