stefandreyer/CODESYS-MQTT #28: Installed Codesys-MQTT Library

fidarin, 2019-11-05:
Hi Stefan, I am interested in your Codesys-MQTT project. Unfortunately, I found these errors after I have installed all libraries from the project's repository. The version of Codesys is Codesys V3.5 SP 15 Patch 1 (32 bit). Is there any problem with the version I am using?
[image]
[image]
[image]

stefandreyer, 2019-11-05:
Hi,
try download missing librarys.
BR Stefan

fidarin, 2019-11-06:
I have installed and added the needed libraries in Codesys, but same errors are still appearing. Is there any other software components that I should have renewed?

stefandreyer, 2019-11-06:
Hi,

there should be a button in you library manager for this purpose, did you try it?

BR Stefan

fidarin, 2019-11-07:
Hi,

I've clicked the "Download Missing Libraries" button. At first, some missing libraries were detected and installed. The number of errors were being reduced from 34 to 29.

I tried the same thing because unfortunately some libraries are still remain undetected and there were still 29 errors. But there aren't any missing library in "Download Missing Libraries" window just like in this photo. Do you have any suggestion?

[image]

stefandreyer, 2019-11-07:
Hi @fidarin 

is this one of my GitHub projects? If so you have to change the PLC into the 32bit version. The Projects are made with 64bit PLC.

[image]

BR Stefan

fidarin, 2019-11-07:
Yes,
[image]

That was wrong of me that I didn't notice the "x64". I sincerely apologize @stefandreyer. I wish I had been more conscious and not going to do it again in the future.

fidarin, 2019-11-08:
Hi Stevan,

I've already used the 64 bit version of Codesys
[image]

But the project still haven't been detected. What should I do?
[image]

stefandreyer, 2019-11-08:
hi @fidarin,

I never had fun with both x32 and x64 versions installed. You still have?

You still have both? 

Please try to have just on on your system.

BR Stefan

fidarin, 2019-11-08:
hi @stefandreyer ,
No, I've uninstalled the x32 version and deleted all the remaining x32 files before I install the x64. But that just happened.

stefandreyer, 2019-11-12:
Hi @fidarin 

uninstalling the 32bit verion allways work for me. 

Try download missing device description, right click on the PLC.

Or changer the PLC to the PLC of you CODESYS Version. Right click on PLC, update device.

BR Stefan

closed 2019-11-26 by stefandreyer