stefandreyer/CODESYS-MQTT #31: 20 unresolved references after download (Crypto related)

MartijnVanAlphen, 2019-12-10:
Dear Stefan,

Trying to run your MQTT library on an Epec 6200 controller. Codesys 3.5.13 (newer is not recommended for now by Epec).

I managed to get all the libraries installed (downloading missing libraries only helped partially). To compile the application all works fine now (I had to add ^ to DATA.PayloadString and StrPt in ReceiveValue.PublishReceived and had to change CommonTypesAndFunctions.FindLastDot to CommonTypesAndFunctions.FindLasDot until I realized that you have an updated version of the library in a different repository, still version 1.0.0.0 according to Codesys though).

So it compiles with no errors. But after downloading I get 20 errors, all Unresolved references related to Crypto. I did install the CmpCrypto library, version 3.5.11.0. Don't know whether it is actually related to this library. Any idea what this could be? Is there a way to work around this?

Out of curiosity: Yesterday you replaced MQTT.library v1.1.0.0 with a different v1.1.0.0. What changed?

[image]

Thanks, Martijn

stefandreyer, 2019-12-10:
Hi Martijn,

old Story,

your PLC Manufactor doesn implement openSSL on your plc.
Doesn't they like security in Finnland? ;) Open a case at Epec, Maybe they implement it...

For your solution look here:

[https://github.com/stefandreyer/CODESYS-MQTT/issues/25#issuecomment-544242203](url)

My Change yesterday:

receive publish with renaining lenght of 128 Bytes will crash calculation of remaining length.

I solved this.
And some other minor Things....
BR Stefan

MartijnVanAlphen, 2019-12-12:
Thanks Stefan! 

I didn't realize this to be the same, since I have no building errors.

I'll give this solution a try and let you know. For sure I'll check with Epec about this!

Martijn

MartijnVanAlphen, 2019-12-12:
Unfortunately not much changed. Moved to 18 unresolved references now, instead of 20. The library CmpTls is completely not there (does give me an error at first, but will let me download anyway).

Can you point me in the direction where these references are used? I can't find any in the libraries nor in the project.

Martijn

stefandreyer, 2019-12-12:
Hi Martijn,

just delete CmpTls library out of commonTypesAndFunction, then compile(check pool objects in build) and you will see the usage.

[image]

BR Stefan

MartijnVanAlphen, 2019-12-21:
It is exactly what I did before. Didn't help.

Martijn

stefandreyer, 2019-12-23:
Hi Martijn,
please try to delete CmpCrypto too…

[image]

BR Stefan

stefandreyer, 2019-12-23:
And it is in the mqtt lib too.

BR stefan

MartijnVanAlphen, 2019-12-23:
Thanks Stefan! I'll try it after holidays. Have a nice Christmas!

Martijn

stefandreyer, 2020-02-05:
Hi Martjin,

did you had luck?

BR Stefan

closed 2020-04-18 by stefandreyer