stefandreyer/CODESYS-MQTT #32: Open project libraries 64 bits codesys windows 10 64 bits

runtimevic, 2020-01-08:
I can't open a project for windows 10 64 bits, through 32 bit libraries, what would be the way to open *.  project with its corresponding libraries for 64 bits?

stefandreyer, 2020-01-08:
Hi,

I'll have a look to remove this libs from the librarys, the usage is small.

Which librarys raise the Errors?

Do you use Twincat or codesys for your plc?

BR Stefan

runtimevic, 2020-01-09:
> Hi,
> 
> I'll have a look to remove this libs from the librarys, the usage is small.
> 
> Which librarys raise the Errors?
> 
> Do you use Twincat or codesys for your plc?
> 
> BR Stefan

Hello, 
.- the libraries that give me error are the ones of oscat that already puts in the store of codesys that the requirements are for 32 bits and that 64 bits is not supported, sysfile23 and cmperrors interfaces.
.-This is for codesys therefore I will see how it works in Codesys, but it would be nice to also have a project in Beckhoff Twincat but I am not able to find a twincat project in mqtt to send and receive multiple subscriptions and publications by topics, if you have a Work done for TwinCAT 3 and can you send me be grateful.
best regards, Víctor.

stefandreyer, 2020-01-20:
Hi Victor,

I'd made my one ports of the OSCAT Libs in my repositiory. I checkt them and didn't found sysfile23 or cmperrors, Only cmperrors2.

Please have a look to [https://github.com/stefandreyer](url) and use the libs from there.

Let me know if it works.

BR Stefan

runtimevic, 2020-01-20:
> Hi Victor,
> I'd made my one ports of the OSCAT Libs in my repositiory. I checkt them and didn't found sysfile23 or cmperrors, Only cmperrors2.
> Please have a look to https://github.com/stefandreyer and use the libs from there.
> Let me know if it works.
> BR Stefan

Hello Stefan,
I downloaded your projects and I was not able to migrate it to a project for 64-bit windows, it gave me bookcase error.
If you get it let me know how you did it and if you can submit a project or put one for 64 bits on the github would be fine.
thank you so much for everything, Victor.

stefandreyer, 2020-01-20:
Hi Victor,

my GitHub Test Projekts are made for and with CODESYS for Windows 64bit.
[https://github.com/stefandreyer/CODESYS-MQTT/blob/master/TestMQTTGithubWindowsWithTLS.project](url)
[image]

Please try to open and run one of my test Projects. If there are erros, Show them.

BR Stefan

runtimevic, 2020-01-21:
> Hi Victor,
> my GitHub Test Projekts are made for and with CODESYS for Windows 64bit.
> https://github.com/stefandreyer/CODESYS-MQTT/blob/master/TestMQTTGithubWindowsWithTLS.project
> 
> Please try to open and run one of my test Projects. If there are erros, Show them.
> BR Stefan

Hi Stefan, this project can be opened and transferred to a w10 64-bit smoothly, all ok.
would have opened another project would be the raspberry.
I am looking for an example project of mqtt from multimessage to multiple publications and for example a subscription that arrive multiple orders, I do not know if in the projects that you have put which would be the most suitable to be able to look at me, the payload would arrive in Json format.
examples would be:

plc subscription
house/commands/-
and they come for example:
topic: house/commands/light1/
payload: '{''cmnd'":"on"}'
topic: house/commands/rgb2/
payload: '{'cmnd'":"off"}'
topic: home/commands/cover1/
payload: '{''cmnd'":"open"}'
and the plc publishes:
topic: home/state/light1/
payload: '{''stat'':"on''}'
topic:home/state/rgb2/
payload: '{''stat'':"off"}'
topic: home/state/cover1/
payload: '{''stat'':"open"}'
Best regards, Víctor.

stefandreyer, 2020-01-21:
Hi Victor,
 I use the lib for raspberry too and it works fine.

the other one is what i acually work on. I use the JSON lib from [https://forge.codesys.com/lib/pro-json/home/Home/](url) to communicate with zigbee2mqtt. I startet coding a library for zigbee devices.
This payload is created by the lib : {''state'':"on''} the lib will be very easy to use and to implement to own projects. There is no need to write any topics because the were createt out of the instance names.

Please give some time for this job.

BR Stefan

stefandreyer, 2020-03-11:
Hi Victor,

done the job, have a look here:
https://github.com/stefandreyer/CODESYS-zigbee2mqtt
BR Stefan

closed 2020-04-18 by stefandreyer