stefandreyer/CODESYS-MQTT #33: Publish Received Method - Pointer to String Error

Araforn, 2020-01-30:
Hi Stefan.

I've been very interested in implementing your library in my project using home assistant and a local broker on my network ( I wont be using an online broker).

I've had numerous library errors but I'm now down to 3 errors on your example project. On the Publish received method I  am getting the following error pointer to string error.

[image]

I can't edit this method so i'm not sure what to do next?.

Do you have any ideas?.

Thanks again.

stefandreyer, 2020-02-02:
Hi Araforn,
did you changed the data types? 

The Example sholud work without any changes…

What were the Errors?

I'll update the example and load it up...

BR Stefan

Araforn, 2020-02-02:
Hi Stefan.

No I didnt change anything?. The errors are as you see above "cannot convert pointer to string" 

I want to use your library for simpe interface with a broker on my network to talk to device in Home Assistant.
I just want to subscribe to 4 topics and be able. To publish when I want to control devices. I know all the topics I need and have tested these in Mqtt  fx. 

I have developed an interface for my home i. Codesys so I need Mqtt functionality in Codesys that can handle a lot of subscription messages. Sometimes I can get 30 at the same time for 1 topic. 

Can your library handle and store a large amount of messages?. Perhaps you have a more simple example?

Thanks so much for making these available.

stefandreyer, 2020-02-04:
Hi Araforn,

yes the lib can handle a lage amount of messeges and this very fast to. My test working with up to 600 messeges/second.

the lib is configt to handle up to 50 subscriptions. This is a Parameter an can be changed. But you can handle as much "Sub subscriptions" as you have RAM on your system(dynamic list and callbacks).

A more Simple example.. I working actually on a libraray for codesys <--> zigbee2mqtt, mabye this makes it easier.

You pointer issue surprises me....

I'm thinking of it.

BR Stefan

stefandreyer, 2020-03-10:
Hi,

please now have a look at

[https://github.com/stefandreyer/CODESYS-zigbee2mqtt](url)

It uses JSON for communicate CODESYS <--> JSON <--> broker <--> JSON <--> zigbee2mqtt.

If you have more questions, feel free...

BR Stefan

Araforn, 2020-03-12:
Hi Stefan.

Thanks for your reply. However I really need a simple MQTT client that will subscribe to a few topics that I have setup in Home Assistant. Opening the Integration example project for CODESYS-zigbee2mqtt this wold not suit my needs.

Basically I have a Codesys project that I developed to interface with Samsung Smart things. 

https://community.smartthings.com/t/raspberry-pi-codesys-project/79308/32

Separately I developed a python based web socket smart app that would write events to the codesys OPC UA server.
https://community.smartthings.com/t/smart-app-multiple-device-subscriptions/159166

This all worked but had limitations.

I since moved to Home Assistant and I discovered MQTT. In home assistant I have "Statestream" setup where it will publish all events from my instance of Home Assistant to MQTT topics. 

https://www.home-assistant.io/integrations/mqtt_statestream/

I need my codesys project to subscribe to these topics  which I will then use these strings to populate all the devices in my codesys project. 

The challenge I have is finding a codesys client that can receive a lot/high rate of published messages and store buffer them in an array for example.

I have tried the below library
[image]

This has a simple interface as below
[image]

Issue is that it cannot handle a lot of fast published events to a subscribed topic.

I would love to use your library but still have the issue with the pointer to string error. Can your library provide what I need like the example above?.

Thanks again,

Paul.

jshroyer031, 2020-03-12:
Paul,

I have the same exact issue and agree that would fix the problem. I also
noticed that you have version 3.5 of the MQTT Client library. Can you send
that to me? I can only find version 3.0.

Thanks,

Jordan

Araforn, 2020-03-12:
Its here:
http://codesys-mqtt-library.sourceforge.net/

What is your exact problem? You have the same problem in stefandreyers library as per this topic?.
Or you have the same problem in Stefan Rossmans MQTT library?.

jshroyer031, 2020-03-12:
Thank you for sending me that library. My exact problem is that when I am
subscribed to the MQTT broker and something is published very large or very
fast... the project crashes. I agree with you that we need to buffer the
incoming data somehow and then parse out what we actually need.  I am not
entirely sure how to do that or if we can do it with the current
library properties.

-Jordan

Araforn, 2020-03-12:
There is a source error in Stefan Rossmans MQTT library when there is high rate of published events on its subscribed topic so its not usable for my application.
Access violation Error
[image]

Source position and subscribed topic (uses a wildcard so it will have a lot of messages published)
[image]

stefandreyer, 2020-03-12:
Hi,

first some infos, my lib is able to handle
a: realy big messageges
b: high periode messeges

The buffer for the messages can be adjust at startup and is created dynamicaly.

One thing in my lib is different to the standard, if the client receives a to big packet the standard wants to diconnect the client. I just throw it away, because, if this is a retained packet, we would never come back to run...

A user of my library did made a nice doku for subscriptions in his project, this as a first hint:
https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/blob/master/docs/FAQ/Howto_adding_a_new_MQTT_subscription.md

Some short details: the subscription fb subscribs to the desired topic(test/#), then if the client receives a packet wich fits to the subscription topics the subscription fb calls the PublishReceived method of the fb passed at the callback input of the subscription FB.

I made a callback collector to so that more than one PublishReceived methods of multiple fbs can be called and the FB itselfs tests if the topic fits to them. Like here:

[image]

This is the big trick so the FBs are able to add themselvs to the collector like here:

[image]

So with the use of the reflection attribute to strings I am able to generet the FB topic out of Instance names.

to your string pointer problems: please look at the declaration of  CALLBACK_DATA, it should look like this:

[image]

and GetValueFromStringPt, should look like:

[image]

If something differs, please get the actual lib.

@Araforn: such a easy FB like you mentioned is sure possible but would brake the libraray. The good things of the callbacks are, if there come 50 packets in one cycle, this 50 packets are handled in this cycle, there comes the speed from.

BR Stefan

Araforn, 2020-03-16:
Hi Stefan,

Thanks for sharing this information.

Its much appreciated. Using the example project by Michiel Vanwelsenaere I dont get any errors and can go online. I have issues understanding his approach for my application so I have posted on an issue as below.

https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/issues/54

Thanks for your help

Araforn, 2020-03-19:
This has been solved in https://github.com/MichielVanwelsenaere/HomeAutomation.CoDeSys3/issues/54

closed 2020-03-19 by Araforn