stefandreyer/CODESYS-MQTT #34: Support for x64 Devices?

rdcoulton, 2020-02-28:
Hi,

I have you done any work to-date to enable the MQTT library to work on 64 PLC devices?

From what I can see its the reliance on SysSocket23 library that is not supported. I have tried to replace it with another that works on 64 PLCs but it just seams to create more errors.

Any thoughts would be appreciated as it worked well on our original system.

Thanks.

stefandreyer, 2020-02-29:
Hi,

the lib works for both 32 and 64 bits. I develop it under 64 bit with 64 bit windows plc and use it with the raspi...

All Windows examples im my repository are for 64 bit, You may have a look at them.

What is your target plc?

BR Stefan

closed 2020-04-18 by stefandreyer

FeLiiix, 2023-12-26:
Hi Stefan, 

I have the same issue. My runtime is on a Proxmox LCX container (Linux).

[image]

I use the latest version of the library.

[image]

stefandreyer, 2024-01-04:
Hi Felix,
what do you use for a PLC? Still I have no Problems in my 64Bit Test Environment on PC.
What is the exact Error? Didn't see it in your Screenshots.
BR Stefan

 
 

image.png (view on web)

image.png (view on web)