stefandreyer/CODESYS-MQTT #35: Publish works from Function Block but how do I subscribe?

jshroyer031, 2020-03-10:
I was able to successfully publish to an MQTT broker using this function block in Codesys running on a raspberry pi but the function block did not have a subscribe option... can anyone help?

stefandreyer, 2020-03-10:
Hi,

yes my docu is poor ;)

subscription are done with the subscription FB wich gets the subscription topic and a callback collector. If the client gets a packet for the subscription all receive methods collectet in the collector are called and the FBs can check if the topic fits to them.

Please look to [https://github.com/stefandreyer/CODESYS-zigbee2mqtt](url) too how this makes things easyer.

More questions? Feel free...

BR Stefan

jshroyer031, 2020-03-10:
Stefan,

Thank you for the response! Great job by the way figuring this all out.
This is wonderful. I cannot open the link that you sent to me
unfortunately. Not sure why.

Thanks,

Jordan

stefandreyer, 2020-03-10:
hi

try this

https://github.com/stefandreyer/CODESYS-zigbee2mqtt

closed 2020-04-18 by stefandreyer