stefandreyer/CODESYS-MQTT #37: Compile Error

ptC7H12, 2020-03-19:
The compiler throws some errors, please have a look at the picture below.  What can I do?
My PLC ist a Wago PFC200.

[image]

stefandreyer, 2020-03-21:
Hi,

geht deutsch?

du musst in deinem Projekteinstellungen den Dynamischen Speicher aktivieren. so 50kB solltne tun.

siehe

https://github.com/stefandreyer/CODESYS-MQTT/issues/18

Grüße Stefan

ptC7H12, 2020-03-21:
Hi,
danke, ich habe jetzt so weit alles provisorisch programmiert, aber irgendwie will der nicht.
Ich schicke Daten an den Broker, und habe den entsprechende Thema in Codesys abonniert, aber ich sehe nirgendwo, dass ich was bekommen habe, genauso wenig wenn ich im Codesys im Publisher das "send" bit auf true setzte, kommt nichts beim Broker an.

Was mache ich falsch, oder was habe ich vergessen?
Das InOut habe ich in die GVL gepackt.
So sieht mein Programm zunächst aus:
[image]

stefandreyer, 2020-03-22:
Hi,

der Inhalt der Empfangenen Daten landet in den FB's die du mit dem collector gesammelt hast. bei den FBs wird die Methode PublishReceived aufgerufen, welche die Daten dann Prüfen kann.

Für einen ersten Eindruck der Daten kannst du einen Breakpoint in publishReceived des collectors setzten und dir unter Data die empfangenen daten anschauen.

Grüße

ptC7H12, 2020-03-22:
Ich bin gerade dabei das mit dem Subscriber zu testen.

Was mir aber gerade auffällt ist:

1. Wenn der Broker nicht erreichbar ist und der Client in den Timeout geht - wie komme ich da wieder raus?

2.  Wenn ich dann z.B. die Adresse des Brokers wechsle, geht der trotzdem nicht in den connected status über, weil er im MQTT_Client zwischen Schritt 10 und 60 wechselt (fehlender Dynamische Speicher). Ich muss dann erst händisch den Schritt auf 0 setzen und den Speicher erhöhen. Dann erst baut er die Verbindung neu auf.
(Der markierte Schritt wechselte zwischen 10 und 60)
[image]

ptC7H12, 2020-03-23:
Ich habe das ganze jetzt mal identisch Aufgebaut, aber bei mir wird das Init vom StringReciver nicht ausgeführt.
[image]

Ich kann zwar mit dem Haltepunkt meine gesendeten Daten sehen, aber dadurch komme ich nicht an die dran, oder?

stefandreyer, 2020-03-29:
Hi,

hmm, klingt komisch. 

Rufst du immer wieder das put des collector auf? Das darfst du nur einmal(Zeitlich), sonnst mach der das solange bis der Dynamische Speciher alle ist, und dann geht der Rest auch nicht mehr. 

Ich mach das immer mit dem Init Bit, wenn nicht gesetzt, put aufrufen und Init bit setzten, so wird das beim Neustart der SPS immer ordentlich verarbeitet.

Wenn es einen Timeout zum Broker gibt sollte er eigentlich reconnecten.

Ja, das mit dem Collector machts sehr einfach. In der Zigbee2MQTT Bibliothekt hab ich das schön ausgereitzt und macht es am Ende sehr Übersichtlich und schön abgetrennt zu dem MQTT Thema.

Noch mehr Erfolg gehapt?

Grüße Stefan

ptC7H12, 2020-03-30:
Habe ich eigentlich schon so gemacht.
[image]

stefandreyer, 2020-03-30:
Kannst du das Projekt mal einstellen?

Grüße

stefandreyer, 2020-04-06:
Noch was erreicht?

ptC7H12, 2020-04-12:
> Noch was erreicht?

Ne, leider nicht,
[Git.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/4466787/Git.zip)

 habe noch mal die CPU neu gestartet, lief aber trotzdem  nicht.

stefandreyer, 2020-04-12:
Hi,

habe mal geschaut. Habe das Projekt auf die Windows SPS umgestellt und so geändert das das läuft. Wenn der publish FB sendet, kommts auch zurück.

Probleme: Falschen receive FB genommen, da muss ich meine Bibliothek aufräumen und dokumentieren.

Bei dem neuen receive FB den compare String gesetzt.
SET_MQTT_IN_OUT wieder ein kommentiert.

Wichtige Frage zu deinem Verbindungsproblem: welchen Broker nutzt du? Es gibt Broker die sich nicht an den Standart halten, und ein Problem damit haben, das eine Paket ID sofort wieder verwendet wird, so wie ich das implementiert habe. Wenn Paket für ID 1 durhc ist, wird die wiederverwendet. Problem kenne ich vom IoBroker MQTT Broker. Diese Broker beenden dan die Verbindung.

Ich habe es mit test.mosquitto.org versucht, läuft.

[Git.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/4467091/Git.zip)

Viel Erfolg!
Grüße

stefandreyer, 2020-04-23:
Läuft?

ptC7H12, 2020-04-25:
Sorry, komme im Moment nicht dazu das zu testen, melde mich, sobald ich das getan habe...

closed 2020-09-09 by stefandreyer