stefandreyer/CODESYS-MQTT #40: Alarm Manager and Trend Storage Compile Errors when using MQTT

Araforn, 2020-04-23:
Hello Stefan,

I am raising this issue as im am trying to implement the library as a follow on from issue 33.
https://github.com/stefandreyer/CODESYS-MQTT/issues/33

I successfully connected a test project and get feedback from my devices.
see below image with realtime status of some my devices.

[image]

However using this same

 project and adding an alarm manager and a trend manager I get the below compile errors.

[image]

I have tried everything I can to resolve (changing library version, changing codesys version etc) but nothing works. They seem to be storage related errors. I have attached the export of the working test project.

If you open it and add an alarm or trend manager and try and compile you will see the errors.

Can you let me know if you are too using an alarm manager or trend manager with any use case of your projects.

Perhaps this is a conflict not yet identified?.

stefandreyer, 2020-04-23:
Hi,

I miss FB_BARDEN_MQTT in your export.

So I can not test..
BR Stefan

Araforn, 2020-04-23:
Apologies Stefan,

Please find now attached with all POU's.

[TEST_MQTT.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/4525448/TEST_MQTT.zip)

stefandreyer, 2020-04-24:
Hi,

I don't use any of the alarm or trend manager. Webvisu don't too.

Please try this. I made a new projekt and copyed your content.

BR Stefan
[Unbenannt3.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/4529725/Unbenannt3.zip)

Araforn, 2020-05-05:
> Hi,
> 
> I don't use any of the alarm or trend manager. Webvisu don't too.
> 
> Please try this. I made a new projekt and copyed your content.
> 
> BR Stefan
> [Unbenannt3.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/4529725/Unbenannt3.zip)

Stefan. 

This worked!. I used this to build and import my project. Its been a few days of work but I have my entire application reworked now using your MQTT library.

Can't thank you enough for your help.

Kind Regards,

Paul.

closed 2020-05-05 by Araforn