stefandreyer/CODESYS-MQTT #41: Topics in GreatExampleOfAdvantages

annD-annD, 2020-06-15:
Dear Stefan,

I tried your great MQTT-library in my Codesys 3.5.16.0 on a PFC200 G2 for my home. I adapted only little references to the recent oscat libraries (little changes because of compatibility with recent codesys version) and your GreatExampleOfAdvantages is running for a few days now. 

I have problems to understand how the topics are created. I see the instance name correct but I think MyTopicSend and MyTopicReceive does not include the "block"name. So only the Lamp1 in the example is working fine by the Topic. I cannot control Lamp2 and Lamp3 and I donot receive changes from them. Do you see a problem in my attached pictures?

Thank you and btw: Ich spreche auch Deutsch ;-)

annD

[image]
[image]

stefandreyer, 2020-06-16:
Hi,

Verrückt, das Problem hatte ich auch schon mal nach einen Update von CodeSys. Bitte Sichere mal das Projekt und lass es mir zukommen, Und dann versuche mal ein "Alles Bereinigen".

Zu dem Fehler: bei dem Instance Name fehllt die Letzte Instance, dadurch wird eine Ebene zu tief das ganze gebildet. Hat mir mal meine Schöne Datenbank zerlegt, da ich da auch aus den namen Schlüssel generriere...

Grüße Stefan

annD-annD, 2020-06-17:
Hi,

danke, dass du es dir ansehen willst. Ich habe anbei ein Projektarchiv exportiert. Es sollte in 3.5.16.0 fehlerfrei übersetzt werden können. Ich habe einen Großteil meiner anderen Programmblöcke entfernt, damit die Dateigröße nicht zu hoch wird.

Viele Grüße,
annD

[Export_von_65_Test_MQTT.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/4794336/Export_von_65_Test_MQTT.zip)

stefandreyer, 2020-07-13:
Hi,

wird noch n bissel dauern, grad viel um die Ohren...

Grüße

stefandreyer, 2020-08-05:
Hi,

entwickelst du mit einem 32bit codesys?

Grüße

annD-annD, 2020-08-05:
Ja, ich nutze bisher ausschließlich die 32bit Version. Lg

stefandreyer, 2020-08-05:
In deinem Projekt schwirren noch 2 32 bit Bibliotheken rum,

Die finde ich gerade nicht..

Grüße

annD-annD, 2020-08-05:
Hmm, welche Namen sind das denn? Ich dachte, ich habe alles eingebettet. Haben diese mit dem PFC200 zu tun?

Viele Grüße

stefandreyer, 2020-08-07:
hi,

diese:
[image]

laufen nur auf meinem 64 bit nicht,

Ich schau nochmal.

Grüße

stefandreyer, 2020-08-07:
Hi,

Problem hat sich hier Bestätigt:
[image]

Im String Instance Name fehlt hinten dran noch ".InstanceName"

Dadurch werden falsche Topics gebildet, kracht aber nur wenn eine Projekt auf 3.5.16.0 gehoben wird, ist n Thema für den Codesys Support. Weis auch nicht wie man das bereinigen kann.

Grüße

stefandreyer, 2020-08-07:
Hier ein Projekt mit 64 bit wo man den Fehler in der Simulation erkennen kann.

[Export_von_65_Test_MQTT bySD.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/5043539/Export_von_65_Test_MQTT.bySD.zip)

Grüße

stefandreyer, 2020-08-12:
Hi,

Problem wurde bei codesys als Fehler beim Wechsel von SP15 auf SP16 angenommen.

Wenn dieser Behoben wird, kann ich nicht sage. Ich werd mal versuchen, ob man das auch anders auf SP16 bekommt.

Deswegen gehen die Topics bei deinem Testprojekt nicht.

Grüße Stefan

annD-annD, 2020-08-12:
Danke für deine Bemühungen.

Lg Andreas

annD-annD, 2020-09-29:
Hi Stefan,
ich habe gesehen, dass es inzwischen die Version 3.5.16.20 zum Downloaden gibt. Wirst du das testen bzw. deine Bibliothek updaten oder soll ich das bei mir mal probieren? Ich möchte nur vermeiden, dass ich dann von dir "überholt" werde;-)
Lg Andreas

stefandreyer, 2020-09-29:
Hi,

werd ich, denke komme die so in 2 Wochen dazu.

Grüße

annD-annD, 2020-10-01:
Hi, danke. Dann werde ich auf deine Infos warten.

Magst du dabei bitte meine Oscat_Network 1351 (in meinem obigen Beitrag vom 17.6. enthalten) verwenden und nach erfolgreichen Test in deinem Repository "OSCAT-NETWORK" austauschen bzw. aktualisieren?

Diese basiert auf der Version 1.35 von 2017 von peewit aus dem Oscat-Forum (die 1.21 aus dem Codesys Store ist von 2013) und es wurden einige Funktionsblöcke aktualisiert bzw. hinzugefügt. Details dazu gibt es hier: [https://drive.google.com/file/d/13yhqqiLaM7E74wRxIVF76SLRpFP6AkvG/view](url)

Damit wären dann mal alle auf Stand und wir können zukünftige Infos und Bugfixes in deiner Version für Codesys 3.5 weiterentwickeln (was leider im Oscat Forum derzeit nicht mehr gewährleistet wird).

Vielen, vielen Dank für deine Mühen,
annD

stefandreyer, 2020-10-01:
Hi,

hab nen ersten Versuch gemacht, scheint mit der neuen Version zu klappen.
Werde mich dann die Tage an die Aktuallisierung machen.
Ich werd noch ne Kobntrollfunktion anlegen, welche das ganze nicht freigeben kann, falls die Namen falsch generiert werden.

Deine Network lib werd ich mal anschauen, sollte das passen, zieh ich die hier mit rein. hab das ja schon mal gemacht, hast du lange für den Port nach V3.5 gebraucht?

Grüße

annD-annD, 2020-10-02:
Hört sich gut an. Das Portieren war jetzt nicht so aufwendig (Namensräume ergänzen, Deklarationen anpassen, übersichtlichere Anordnungen usw.) weil grundsätzlich keine funktionellen Änderungen notwendig waren. Aber ich finde es schade, dass es im Moment keine einheitliche Quelle für den aktuellen Stand der V3 Bibliothek gibt, wo zukünftige Anmerkungen dann auch einfließen können.

Viele Grüße

stefandreyer, 2020-11-02:
Hi annD,

zieht sich hier gerade etwas, hab ne Baustelle zu Hause, wird bald weiter gehen.

Grüße Stefan

annD-annD, 2020-11-03:
Danke Stefan,

dein Beispiel läuft schon mal stabil bei mir. Ich muss mir aber noch überlegen, wie ich das am besten bei mir einbinde, damit ich nicht den Überblick verliere. Ich möchte ja einige schon vorhanden Schaltzustände, Dimmwerte oder andere vorhanden Daten per MQTT mit ioBroker und Tasmota-Geräten austauschen. Die Blöcke von MichielVanwelsenaere habe ich mir dazu auch angesehen, aber diese machen mir auch einen sehr komplexen Eindruck.

Schöne Grüße,
annD

stefandreyer, 2020-12-04:
Hi,

Habe jetzt Version 3.5.16.30 online gestellt.
Codesys Network ist jetzt auch auf Version 1.35. Aber hier Vorsicht. Deine Version konnte ich nicht übernehmen, da die 32 bit Bibliotheken die du genommen hast mit den 64 Runtime Systemen kollidieren. Musste also noch was ändern und Strechen. Werde das aber noch Versuchen gerade zu ziehen.

Wegen komplexität: schau dir mal mein Beisbiel für Zigbee2Mqtt an. Erschreckt einen Vielleicht am Anfang, aber am Ende ist es total Einfach. Nur eine Instanz eines FBs erstellen und Topics und alles weitere dir Automatisch generiert.

Wie ist da dein Stand?

Grüße

annD-annD, 2020-12-04:
Hi,

ich habe deine Mqtt-Bibliothek jetzt auch in der aktuellen 64bit Version am
Laufen und in Oscat ein paar Verweise getauscht. Ich nutze deine
Temperatur-Base zum Senden von der SPS an Mosquitto und dann ioBroker.

Ich blicke aber noch nicht ganz durch was ich z.B. ändern muss um die Werte
öfter zu aktualisieren (wenn es statt Temperatur ein anderer Wert ist).

Ich stelle mir vor, für jeden Wert so einen Base Baustein mit automatischen
Topics zu nehmen.

Lg annD

stefandreyer, 2020-12-05:
Hi,

ich werd dafür mal noch n schönes Beispiel entwerfen.
Da Werd ich die einzelnen Themen noch etwas entzerren.

Grüße

stefandreyer, 2021-02-09:
Hi,

ich muss hier nochmal bei...

Grüße

closed 2022-11-22 by stefandreyer