stefandreyer/CODESYS-MQTT #42: RemainingLength liefert 0 zurück und MQTT fängt sich danach nicht wieder.

SvenJoedicke, 2020-07-13:
Hallo Herr Dreyer,

Ich habe folgendes Problem. Ich habe in einem Objekt einen Request-Kanal (MQTTPublish) und einen Response-Kanal (MQTTSubcribe) aufgesetzt. Sobald ich einen Request absetze, bekomme ich von der Gegenstelle einen Response. Dieses Prinzip funktioniert soweit ganz gut. Aber leider nur 3 mal. Ab den 4. mal liefert die FUNCTION GetRemainingLength bei RemainingLength eine 0  zurück, obwohl ein valider Payload im Buffer ist. Was mir aber aufgefallen ist, dass lediglich der Header des Buffers leicht differiert, was zu dieser zurückgegebenen 0 führt. Ich habe leider keine Ahnung, was diese Werte in diesem Header bedeuten. Siehe folgenden Abbildungen.

Gutfall:
[image]

Schlechtfall:
[image]

In beiden Fällen ist zu erkennen, dass der Payload ein und der selbe ist.

Im Schlechtfall sieht man, dass die BufferSize statt 258, dann 259 ist, dass das zweite und das dritte Byte im Buffer [128, 2] ist, anstatt [255,1]. 

Leider weiß ich noch nicht, was diese Header-Werte im einzelnen bedeuten. Es wäre schön, wenn sie Herr Dreyer da Licht ins Dunkel bringen könnten.

MFG Sven Jödicke

stefandreyer, 2020-07-13:
Hallo Herr Jödicke,

kommt mir bekannt vor:

[https://github.com/stefandreyer/CODESYS-MQTT/pull/30](url)

Ich hatte mal genau diese Problem beim überschreiten der Länge 255, genau wie bei Ihnen.

Bitte Prüfen sie ob sie die aktuelle Version verwenden.

Mit freundlichen Grüßen

Stefan Dreyer

SvenJoedicke, 2020-07-13:
Wie prüfe ich das??? Ich hatte in der Bibliotheksinformation reingeschaut
und da steht 1.1.0.0... das müsste ja aktuell sein, oder???

Mfg
Sven Jödicke

stefandreyer, 2020-07-13:
Auf GitHub liegt aktuell 1.1.0.2:

[image]

SvenJoedicke, 2020-07-13:
Hallo,

Soll das generell bedeuten, das subscribte Nachrichten über 255 Probleme
bereiten??? Das würde mich wundern, ich habe schon weit aus größere
Nachrichten erhalten, so um die 1000 Zeichen...

Mfg
Sven Jödicke

SvenJoedicke, 2020-07-13:
Hallo,

Ich muss es morgen mal prüfen auf Arbeit...

Mit freundlichen Grüßen
Sven Jödicke

stefandreyer, 2020-07-13:
Hallo,

Nein kein generelles Problem über 255.

Das war nur ein Problem knapp über denn 255, wenns noch größer wird gings wieder...

War wegen der dynamischen Erweiterung der mitgeführten Bytes für die länge...

Mit freundlichen Grüßen

SvenJoedicke, 2020-07-13:
Okay, vielen Dank ... Prüfe ich morgen...

SvenJoedicke, 2020-07-14:
Vielen Dank, hat funktioniert mit dem Update auf Version 1.1.0.2

closed 2020-07-14 by SvenJoedicke

SvenJoedicke, 2020-07-14:
Hallo Herr Dreyer,

meine Firma und ich möchten uns wegen dem guten Support gern erkenntlich
zeigen. Gibt es eine Möglichkeit Ihnen eine Spende zukommen zun lassen, per
Paypal zb. ?

Mit freundlichen Grüßen
Sven Jödicke