stefandreyer/CODESYS-MQTT #44: Payload size only 80 characters for SD_MQTT.CallbackCollector and SD_MQTT.ListStringCallback

SilviuMarchidanu, 2020-07-18:
Hi Stefan , 
I have a little problem with the payload size for the SD_MQTT.CallbackCollector , thru receiver :	SD_MQTT.ReciveString;  , also with  SD_MQTT.ListStringCallback; I get only 80 Characters even thou the payload strings are SET for 255 characters .
NOW with SD_MQTT.SingelStringCallback;  I get up to 2000 characters on the payload string , as that is the max config for the handier.  

Can you advice on the problem ?

stefandreyer, 2020-07-18:
Hi,

this should work, i'll have a look on this....

BR Stefan

SilviuMarchidanu, 2020-07-24:
Thanks Stefan , looking forward .

stefandreyer, 2020-08-05:
Which Lib version do you use?

stefandreyer, 2020-09-09:
Hi,

still these problems?

SilviuMarchidanu, 2020-09-22:
Hi Stefan , 
Apologizes for the delay , been traveling a lot with work .
The first time I have encountered the problem was with library version 1.0.0.0 as that project is developed under Sp12  .

Then under new version 1.1.0.0 I have tried with the PublishReceived and ReceiveString FB , same result just 80 Characters , I 'm trying to develop my own ZigBee FB Devices with Json and trying to develop under SP 16 .

Let me know if I'm skipping something .

stefandreyer, 2020-09-22:
Hi,

do you have a example projekt for me? Or a screenshot where I can see this.

You ma look to my zigbee2mqttDevices project. There are allready some devices and easy to extend and to implement. But be aware, under SP16 there is a bug, so the instance names are not beeing determinated korrect.

BR Stefan

closed 2022-11-22 by stefandreyer