stefandreyer/CODESYS-MQTT #45: RTC_C output VAR LDT gives error

warwolf7, 2020-07-23:
Hi, in the timeDate\RTC_2  FB, the VAR_OUTPUT LDT is giving me an error. Can this be happenning because of the LDT data type already in  codesys?
https://help.codesys.com/api-content/2/codesys/3.5.16.0/en/_cds_datatype_date_and_time_of_day/

Should this variable be renamed?

warwolf7, 2020-07-23:
it's an error that comes up with the compiler 3.5.16.0. It does not happens with compiler 3.5.12.0

annD-annD, 2020-07-23:
You are right. Here is some information:
[http://www.oscat.de/community/index.php/topic,5794.msg18140.html](url)

stefandreyer, 2020-08-10:
Hi,

I Update my libs allready, need to publish them...

BR Stefan

Tommilein, 2020-08-11:
Hallo Stefan,
we had a problem with oscat basic in July 2019 already:
[image]

I hope it helps to remember because it sounds like the current.

And now I have the same problems like warwolf7. I made an upgrade to 3.5.16 and nothing functions.
[image]

I hope you can publish the new libraries in a short time. Your library is in production (smile) !! 

beste Grüße
Thomas

stefandreyer, 2020-09-09:
Hi,

libs are up to date, please have a look.

BR Stefan

Tommilein, 2020-09-14:
Hallo,
I tried the Lib and got 3 Errors. I changed the MQTT-Lib only and now I have two different versions of OSCAT-Basic in the library manager.
[image]

Is it enough to change to the MQTT-Lib "MQTT 1.1.0.3.library"  only?

bye
Thomas

stefandreyer, 2020-09-14:
Hi Thomas,

try to remove your version of OSCAT Basic of the repositiory and to work with my only.

I think this should help.

BR Stefan

Tommilein, 2020-09-14:
Hi Stefan,
the OSCAT-BASIC-Lib is not in the zip-File. Where can i find it or must i export the Lib from MQTT-Lib?
I have problems to understand the library manager.

Thomas

stefandreyer, 2020-09-14:
Hi,

sounds good...

have fun!

closed 2020-09-14 by stefandreyer