stefandreyer/CODESYS-MQTT #48: Cannot Find My desired retrived Data

diditsub, 2020-11-03:
Hello Dreyer,

I'm Using your newest MQTT Library Ver 1.1.03 on Wago PLC PFC2000.
My Schemes Are Sending Data From Omron PLC (Publish) to my PFC2000 PLC (Subscribe) Through MQTT with mosquitto as my mqtt broker.
I have been successfully connected to both PLC, it is indicates from the mosquitto side.
But i cannot find my data on my PLC Wago, it should be on this picture Below right?

[image]

the data is retrived to my PFC2000 but i cannot find a way to see it and move it to my own variable (Data is null like the picture above).after i open the receiver and put breakpoint into it, the data exist on 'PayloadString^' like this picture below :

[image]

my question is , how do i get that received data and move to my own variable?im new to codesys, im expecting your answer.

*PS: i have been searching the same issue on your post and i found same similar case with me, but the case is closed and i still dont understand how to get my var...

SanderNID, 2020-11-06:
Hi,

Try looking at the HowTo project. This has the perfect example what you are looking for.
In the example below you see that an init function is used which sets some values.
The received string is collected in receiver.MQTTReceived.
[image]

stefandreyer, 2020-11-23:
hi diditsub,

did you had sucsess?

BR Stefan

closed 2020-11-30 by stefandreyer