stefandreyer/CODESYS-MQTT #49: TLS not working

SanderNID, 2020-11-06:
Hi,

I downloaded version 3.5.15 and upgraded it to 3.5.16 and used the project: Interation HowTo.project.
I use the free hivemq open broker for testing purpose (https://www.hivemq.com/public-mqtt-broker/).

First problem:
I tried to connect to "broker.hivemq.com", where TLS is false. This works the 'connected' value becomes true.

Thereafter set the TLS to true and got instant pointer error (in IP_CONTROL [From Common Helper for development]): 
[image]

To solve this problem I added the pointer to tls_config, which was previously pointing to 0.
Although the exception is gone, the connection is still being refused.

[image]

I added the error history. Hope you know what is going wrong, and which values I have to use for tlsMethod, chiperList and verifymode.
[image]

stefandreyer, 2020-11-07:
Hi Sander,

does hivemq provide a tls access to there test broker? You need a different port for use tls. see mosquitto for example:

[https://test.mosquitto.org/](url)

Thats why my tls test project work with test.mosquitto.org.

BR Stefan

SanderNID, 2020-11-09:
Hi Stefan,

Where am I able to change the port number? I know STD uses 1883 and TLS mostly uses 8883, but I didn't find any variable where I can change the port number.

Furthermore, is it the correct way I did by adding the pointer to my own TLS_CONFIG? 

I tried making connection to "test.mosquitto.org" via STD (works) and via TLS, but that still does not work. (could you make a screenshot with example?)
[image]

stefandreyer, 2020-11-09:
Hi Sander,

Port is given by the server string:
[image]

The pointer to your own TLS var by CommonTypesAndFunctions.TLS_CONFIG; is ok.

You may have a lookt into my TestOutOfTheBoxTLS FB in My Lib, there you see the handling.

[image]

I did a test now and had no luck with port 8883 on hivemq.

Good luck.

BR Stefan

SanderNID, 2020-11-09:
Thanks a lot!

closed 2020-11-09 by SanderNID