stefandreyer/CODESYS-MQTT #50: Basic library version issue

SaadSel, 2021-01-04:
Hi stefan

I'm new to PLc programing, I installed the latest version of your MQTT library, but I have an issue with BASIC library ( PS : I installed ther Basic library in ur repos) 

[image]

Thnaks in advance.

stefandreyer, 2021-01-08:
Hi SaadSel,

this is the Basic version from codesys, I installed CODESYS 3.5.16.30. There I did get this Version.

You may try Download missing Librarays.

Which Codesys Version do you run?

BR Stefan

SaadSel, 2021-01-08:
Thanks for the reply.

this is the version i use ( I tried both x64 and x32) : 
[image]

this is what happen when i try to download missing library : 
[image]

stefandreyer, 2021-01-09:
This is strange,

but acually I work an a improved library. There i wand to use placeholders for such librarys. Then those problems should be less. please be patient, I think I'll release it next week.

BR Stefan

SaadSel, 2021-01-09:
Thanks a lot Stefan. I will be waiting for the release of the new version for test

Westliff, 2021-01-24:
Hallo Steffan,

erst einmal vielen Dank fur die ganzen tollen Beispiele, gibt es Neuigkeiten zur neuen Version ? Ich habe den selben Fehler wie SaadSel. 

Gruß
Matthias

stefandreyer, 2021-01-26:
Hi,

I think this will happens in the next days.

Hallo Westliff, OSCAT BASIC hat diesen Versionstand durch eine neue Version von 3S selber bekommen. Die haben noch was wegen den 64bit datentypen gemacht.

Grüße

stefandreyer, 2021-02-09:
Hi,

sorry did take more time than I have ;)

new version is online, you shold be able to change the lib to your version, I did set a placholder for them.

Hallo, 

ich habs nun umgesetzt. die Problematischen Bibliotheken haben nun einen Platzhalter, somit kannst du sie auf deine Version umlenken.

Grüße

Westliff, 2021-02-12:
Hallo Stefan,

Vielen Dank für die Umsetzung :-)

es klappt auch alles soweit nur einen blöden Fehler werde ich nicht los.

[image]

Gruß
Matthias

stefandreyer, 2021-02-12:
Hi,

klingt ja super!

zu deinem Fehler, da hat sich eine kleine Inkonsitens eingeschlichen beim Wechsel zu SP16, 3S hat hier datentypen eingeführt, die vorher als Variablennamen durch OSCAT genutzt wurden. Schon krachts.

Also, entweder auf die aktuelle OSCAT BASIC wechseln, oder hier den Unterstrich in deiner OSCAT BASIC entfernen:

[image]

Grüße

Westliff, 2021-02-16:
Hi vielen Dank,

denke der eigentliche Fehler ist behoben bekomme das Programm Fehlerfrei geladen. 

der MQTT Client scheint auch zu leben allerdings bekomme ich den Public FB in der Demo scheinbar nicht ans Leben. 

[image]

Gruß
Matthias

stefandreyer, 2021-02-16:
Hi,

kannst du dein Projekt minimieren und posten?

Grüße

Westliff, 2021-02-16:
Hallo ja kein Problem.

[MQTT.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/5991148/MQTT.zip)

Gruß
Matthias

stefandreyer, 2021-02-16:
Hi,

schau ich mir mal an. Hatte es schon mal auf. CFC ist nicht so meine Welt...

Grüße

annD-annD, 2021-02-16:
Hi Stefan,

noch mal wegen den Versionen von der Basic. Wenn ich es auf der 3S Homepage herunterlade, habe ich dort nach wie vor die Version 3.3.3.0. Wie bzw. wo kommt es zu dieser Version 3.5.16.0 mit geänderten Variablen? Deine Bibliotheken Building und Network zeigen auf diese 3.5.16.0 welche bei mir aber nicht nachgeladen werden kann. Ich müsste die Verweise darin manuell ändern.

Viele Grüße

stefandreyer, 2021-02-16:
Hi,

die kamen bei mir mit SP16 Patch 3 mit.

[image]

Grüße

stefandreyer, 2021-02-16:
Ich habe die auch schon mal raus gezogen,

weiß aber nicht ob ich die hier posten darf....

Grüße

annD-annD, 2021-02-16:
Waren diese dann im Package Manager schon drinnen? Ich habe gerade 3.5 SP16 Patch 4+(64bit) installiert und danach alle Bibliotheken von dir ergänzt. Diese Basic ist nicht dabei bzw. ich kann sie auch nicht nachladen. Auf der Webseite ist sie offensichtlich separat auch nicht verfügbar.

Grüße

annD-annD, 2021-02-16:
Offensichtlich ist ein neues Release 3.3.4 von Oscat angekündigt:

[http://www.oscat.de/community/index.php/topic,5794.msg18215.html#msg18215](url)

stefandreyer, 2021-02-22:
Hallo,

na das wird ja spannend. 3S hat eigene Namen verwendet. das scheint jetzt mit denen von OSCAT zu kollidieren.

Grüße

hilddav, 2021-02-25:
Hallo Stefan,

leider habe ich auch das Problem mit der Oscat Version 3.5.15.0

[image]

Ich nutze die MQTT Library 1.2.0.2 mit CodeSys (Eaton) V3.5.16 BF1

Nun habe ich schon eine Menge ausprobiert, aber leider bekomme ich das Problem nicht gelöst. Vielleicht hast du oder habt ihr ja auch noch eine Idee was man machen könnte.

Viele Grüße und ganz großen Dank für deinen Einsatz bei der Entwicklung!
David

annD-annD, 2021-02-26:
Hallo,

offensichtlich wurde von 3S jetzt eine Oscat_Basic in der Version 3.3.4.0 (was sich aber mit dem zukünftig angekündigten Release von Oscat überlappt) zum Download bereitgestellt, wo diese Variablen geändert wurden: LDT, LDATE, LTOD to LOCAL_DT, LOCAL_DATE, LOCAL_TOD. Das sollten wir so hinnehmen und in den anderen Bibliotheken anpassen (z.B. auch in der CommonTypes...). Ich habe es bei mir in der Oscat_Network und Oscat_Building auch so übernommen, damit sie sich fehlerfrei übersetzen lassen. Ich hänge beide gleich mal hier an.

@stefandreyer Leider kann ich deine MQTT 1.2.0.2 nicht ohne Fehler übersetzen:
[image]

Der _LDT-Fehler erfordert ein Umbenennen in der CommonTypes... , aber bei den anderen Fehlern blicke ich nicht durch.

Danke und viele Grüße,
annD

[network_1352_building_1002.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/6052688/network_1352_building_1002.zip)

stefandreyer, 2021-02-27:
Hi,

ja hier kollidiert gerade einiges. Ich lad dann mal ne neue Version hoch, auch bei der Common, in der ich mal die Namensräume gerade gezogen habe. 

Ich schaue auch mal in meiner Basic die von 3S kam, nicht das ich da was rein gepfuscht habe ... 

Wenn ja Lager ich das aus.  

Grüße

stefandreyer, 2021-02-27:
Ja, da sind ein paar sachen,

mache gerade eine BASIC Extension LIB, da packe ich das rein.

Grüße

stefandreyer, 2021-02-28:
Guten Abend,

habe jetzt alle Bibliotheken auf 3.3.4.0 umgebogen. 3.5.16.0 ist nicht mehr auf meinem System installiert und läßt sich alles übersetzten. Deswegen musste ich eine weiter Bibliothek erzeugen, welche Funktionen enthält die in 3.5.16.0 da waren und in 3.3.4.0 nicht mehr... Aber so geht das mit der bei 3S verfügbaren Bibliothek. Dazu muss auch eine Andere Version der OSCAT_Network genommen werden, liegt auch bei mir.

Viel Spaß beim testen, gerne Rückmeldung wenns Funktioniert.

Grüße!

hilddav, 2021-02-28:
Hallo Stefan,

vielen Dank, dass du dir das noch mal angeschaut hast. Leider hatte ich noch keine Zeit das in Ruhe zu testen, aber ich versuche es morgen und gebe dann direkt Rückmeldung.

Viele Grüße und noch einen schönen Abend!

annD-annD, 2021-02-28:
Hallo Stefan,

danke für deine Mühe. Kann ich deine Basic_Extension auch irgendwo finden?

Vielen Grüße

stefandreyer, 2021-02-28:
Hi,

Sorry, liegt jetzt im OSCAT Basic repository....

Grüße

Westliff, 2021-03-01:
Hallo Stefan,

Danke für das umsetzen und die Arbeit die du investierst, die Bibiotheken ich habe gerade mal alle geladen und bekomme das Projekt von mir, mit deinem MQTT Client fehlerfrei übersetzt.

Grüße

stefandreyer, 2021-03-01:
Hi annD

Danke für den wink mit der 3.3.4.0 von Oscat, so konnte ich wieder auf einen halbwegs vernünftigen Stand kommen. Es wird aber wieder eine 3.5.16.x geben, da in der 3.3.4.0 LREAL_TO_STRF fehlt. Da hat3S die falsche Basis für genommen.
Dazu In deine Network und Building zu schauen bin ich leider nicht gekommen.

Klappt das so alles bei dir?

@Westliff kein Ding, man gibt halt was an Open Source zurück, was man auch selber nutzt. Meine Art des Spendens ;)

Grüße

annD-annD, 2021-03-01:
Hi Stefan,

ich konnte die Fehler "IPC.ResultReceive" und "IPC.ResultSend" noch nicht beheben. Das muss ich mir morgen nochmal anschauen. Was sind die Unterschiede im IP_CONTROL von deiner Common zur normalen Oscat_Network?

Viele Grüße

stefandreyer, 2021-03-01:
Hi annD

Meiner ist TLS fähig.

Deswegen ist der nochmal in der common drinnen.

Nehm Bitte die Network 1.3.5.2 von mir, ich habe da in der Struktur IP_C rumgeschmiert, hätte ich vielleicht erweitern sollen:
[image]

Naja, ich brauche ja einen Grund für das nächste update:

Grüße

annD-annD, 2021-03-01:
Danke, für den Hinweis. Ich kann jetzt deine MQTT und die Common fehlerfrei übersetzen - aber bei deiner Network schaffe ich das nicht:
[image]

Schöne Grüße

stefandreyer, 2021-03-01:
Hi,

ja, das kann sein... da bin ich noch nicht Fertig geworden ...

Ich denke das sollten ich/wir mal zusammenführen...

Also meine Network für MQTT deine für den Rest...

Grüße

annD-annD, 2021-03-01:
... vielleicht sollte ich nur die geänderte Struktur in meine Network übertragen (wenn sonst keine Änderungen relevant sind). Diese kann ich nämlich direkt übersetzen.

Grüße

stefandreyer, 2021-03-01:
Mach das mal,

dann hier rein, und dann teste ich mal. Wenn das passt, dann lade ich sie hoch.

Grüße

annD-annD, 2021-03-01:
Hier habe ich nur diese IPC Struktur ergänzt (und mal auf Version 1.3.5.3 erhöht):

[oscat_network_1353.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/6064063/oscat_network_1353.zip)

annD-annD, 2021-03-01:
... es wäre schön wenn die Oscat_Basic u. Network möglichst unverändert bleiben, damit im Falle eines Updates seitens Oscat das Nachziehen in Codesys 3.5 keinen großen Aufwand macht. Alle Erweiterungen wären optimal in einer deiner Bibliotheken (z.B. Common).

Danke und viele Grüße,
annD

stefandreyer, 2021-03-01:
Hi, 

alles klar, ich werd mich mal ran machen.

Vielen Dank!

stefandreyer, 2021-03-01:
Das mit dem Ändern sehe ich auch so.

Ich versuch mal die Struktur ausserhalb von der Network zu platzieren. sollte per extends klappen.

Grüße

Westliff, 2021-03-01:
Hallo Stefan,

sagt dir "Dereference of invaild pointer" etwas ?

[image]

Grüße

stefandreyer, 2021-03-02:
Hi,

ja, da wird ist für MQTT_IN_OUT keine gültige Refererenz(Pointer) vorhanden. Das muss mit SetMqttInOut() gemacht werden.  Deswegen verläßt er den FB in Zeile 3 damit es keine Laufzeitfehler gibt.

Grüße

Westliff, 2021-03-02:
> Hi,
> 
> ja, da wird ist für MQTT_IN_OUT keine gültige Refererenz(Pointer) vorhanden. Das muss mit SetMqttInOut() gemacht werden. Deswegen verläßt er den FB in Zeile 3 damit es keine Laufzeitfehler gibt.
> 
> Grüße

Hi Stefan,

kannst du mir auf die Sprünge helfen wo ich das Eingeben muss stehe gerade was auf dem Schlauch. 

Grüße

stefandreyer, 2021-03-02:
Hi,

für dein Publisher muss es sowas geben:

[image]

sub1 gegen birthMessage ersetzten

Grüße

stefandreyer, 2021-03-03:
Hi annD

Die 1.3.5.3 Network von dir ist die Erste auf meinem System die sich sober übersetzten läßt ;)

Ich habe die IP_C Struktur wieder zurück geändert und in der Common erweitert. somit kann die Network unverändert bleiben.

Ich werd noch ein Paar testes machen, die Versionen bereinigen und dann Veröffentlichen. im MQTT Client habe ich auch noch was geändert, das sich die dynamische Speicher allokierung nicht mehr so beschäftigt, wenn der Broker mal abschmiert.

Grüße

Westliff, 2021-03-04:
Hallo Stefan,

Ich bekomme alles fehlerfrei übersetzt der Client an sich verbindet sich auch und schickt die "will" Message an den Broker allerdings hat der Publish FB selbst keine Lust zu arbeiten. Ich kenne mich selbst zu wenig mit den referenzen aus. die Ref für die Birthmessage habe ich ich habe die FB`s aus dem beispiel kopiert bekomme aber immernoch keine referenzen. 
Ist jetzt mein letzter versuch sonst hänge ich das MQTT mit Codesys an den Nagel ;-) 

[image]

Grüße

Westliff, 2021-03-04:
Hallo Stefan,

nochmal vielen dank es Funktioniert jetzt alles. In dem Projekt > GreatExampleOfAdvantagesCFC habe ich nur zwei kleine änderungen vornehmen müssen.

für die birth Message fehlte der FB in der MQTT_Stuff_init
[image]

und für das anschließende Publish eine verbindung des connectRisingQ Inputs im MQTT_Stuff
[image]

jetzt geht es ans probieren mit deiner Struktur die Sonoff/Tasmota switche einzubinden.

Dank & Gruß

stefandreyer, 2021-03-04:
Hi,

ja sorry, ich habe die Beispiele noch nicht an de neue Version anpassen können. Weiter ist mir das CFC mittlerweile ein Graus. Ich mache eigentlich nur noch ST. Ich freue mich aber sehr, das es bei dir jetzt funktioniert. Falls du noch Probleme mit Tasmota hast, kannst du ruhig nochmal fragen. Ich selber nutzte Zigbee2MQTT. das läuft sehr gut.

Grüße

Westliff, 2021-03-05:
Hi Stefan,

hast du evtl nen Tip für mich wie ich das letzte Topic POWER am besten anhänge ?

Würde die SONOFF (jetzt Stern) in die Räume verteilen allerdings ist ja dann das Topic (Stern) die letzte ebene beim MQTT und Tasmota schickt noch ein POWER hinterher. 

[image]

Gruß

stefandreyer, 2021-03-05:
Hi,

ganz kann ich noch nicth folgend, denke aber das ich hier ein ähnliches Thema habe. iobroker setzt ein Set an das objekt um es zu setzten. 
Deswegen habe ich 2 Strings:
[image]
Erledigen tu ich das in einer FB_init Methode mit:
[image]

Solltest du meinen das du einen weiteren Subtopic meinst, würde ich einen FB dafür machen, der das mit reflection mit bringt, oder das in einer FB_Init methode anhängen.

Grüße

stefandreyer, 2021-03-07:
Hi annD,

müsste jetzt so passen. die Network ist Orginal belassen, nur auf V3 portiert. IP_C_EXT habe ich in Common ausgelagert.

Grüße

Westliff, 2021-03-07:
> Hi,
>
> ganz kann ich noch nicth folgend, denke aber das ich hier ein ähnliches Thema habe. iobroker setzt ein Set an das objekt um es zu setzten.
> Deswegen habe ich 2 Strings:
> [image]
> Erledigen tu ich [...]

Hi Stefan,

vielen Dank nochmal habe jetzt alle SonOff am laufen. 👍

Gruß

stefandreyer, 2021-03-07:
Hi @Westliff,

freut mich zu hören. Viel Spaß noch!

Grüße!

annD-annD, 2021-03-08:
Hi Stefan,

Tip-Top. Lässt sich jetzt auch bei mir einwandfrei übersetzen (Software und Bibliotheken aus dem 3S-Store: Codesys 3.5.16.40, Oscat Basic 3.3.4.0 und jetzt auch Oscat Network 1.3.5.2). Und dazu dann deine MQTT 1.2.0.4, deine Basic_Extension 1.0.0.0 und deine CommonTypesAndFunctions 1.1.0.4.

Kannst du bitte bei dir kurz prüfen, ob die Network 1.3.5.2. aus dem Store bei dir auch läuft und ggf. die Version 1.3.5.3 dann ersatzlos löschen? Der Unterschied sollte nur die erweiterte IP_C Struktur gewesen sein, die ich in meiner Version an 3S nicht mitgesendet habe.

PS: Bei den Bibliotheken ist mir noch aufgefallen, dass ich bei der PRO_JSON eine höhere Version 1.0.0.9 hatte.

Mit diesen Infos sollte es auf jeder Neuinstallation sofort lauffähig sein.

Viele Grüße,
annD

Westliff, 2021-03-10:
Hi Steffan,

hast du noch nen Tip für mich wie ich den String aus dem SubTopic am besten mit deiner MQTT.libary auslesen kann ?

[image]

Danke

stefandreyer, 2021-03-10:
Hi annnD

jetzt weis ich wieder, warum die Network bei mir nicht fertig geworden ist:
[image]
SysFile23 und SysSocke23 funktionieren nicht auf 64Bit System wie Windows. Da müssen wir nochmal bei. Das muss wohl alles auf die SysSocket2 Bibliothek umgebogen werden. Doch da ändert sich einiges. Hast du ne andere Idee?

@Westliff 
Ja habe ich. Das was du da hast ist JSON. In meinem Profil findest du eine JSON Bibliothek. Die nutze ich in meiner ZIGBEE Bibliothek. Ist total cool. Mann braucht nur die Struktur  im Codesys nachbauen, und der FB füllt oder liest JSON String <> Objekte.

Grüße

annD-annD, 2021-03-10:
Ich habe bei mir einen Rechner mit Windows 10 Pro 20H2 in 64-Bit Ausführung mit dem Codesys 3.5.16.40 (64-bit) und diese Bibliotheken:

[image]

Ich kann den Fehler nicht nachvollziehen. Hmm?

stefandreyer, 2021-03-10:
Hi annD,

Zielsystem muss 64bit sein(Windows)

Versuch mal den SimpelstReceiver aus meinem Repo.

Grüße

annD-annD, 2021-03-10:
Ja, mit dem Control_Win_V3_x64 erhalte ich dieselbe Meldung. Mit dem Control_Win_V3 und mit meinem hauptsächlich genutzten Control for PFC200 SL funktioniert es (weil offensichtlich 32-bit).

Viele Grüße.

annD-annD, 2021-03-10:
... sind 64-bit Zielsysteme derzeit praktisch soweit relevant, dass es unbedingt berücksichtigt werden muss? Ich kann das nicht abschätzen. Ich bin der Meinung, dass ausführbare Steuerungen (abgesehen von PCs) vom nutzbaren RAM sind unbedingt 64-bit nutzen müssen, oder?

stefandreyer, 2021-03-10:
Hi,

naja schwer zu sage. Ich nutzes es Hautpsächlich zum Testen direkt auf der Windows Maschine. Das es wohl wenig Anwendungen gibt die unbedingt 64bit brauchen gebe ich dir recht. Aber vielleicht kennen wir die einfach nicht. 
Es muss ja nicht unbedingt wegen RAM 64Bit sein. Motion wird damit auch schneller und präziser.
Für mein Gefühl sollte es auf einer ganz Normalen Windows 64Bit Installation einfach lauffähig sein.
Grüße

[1 more reply]

closed 2022-11-22 by stefandreyer