stefandreyer/CODESYS-MQTT #52: Access violation on reset

stefandreyer, 2021-02-11:
This happpends:

[image]

Im working on it...

BR Stefan

stefandreyer, 2021-02-15:
Hi,

one update, crash only happends if MQTT_IN_OUT and so clientFB resists in a GVL.

Looks like a Runtime problem of CODESYS, more investigation needed.

BR

stefandreyer, 2021-02-16:
Hi,

did made a ticket at CODESYS, problem raises when __delete is called in FB_Exit and the instance is in a GVL.

BR Stefan

stefandreyer, 2021-02-16:
Hi,

bug was confirmed by CODESYS. Workaround is available.

I'll see to implement it in the next version, If someone needs it bevor, please raise a hand.

BR Stefan

stefanroelofs, 2021-07-20:
Hi goodmorning,

We are testing your MQTT library in our PLC software, and the same error message appears when we do a reset. 
You say that you have a workaround available? Could you please share that?

We're on CODESYS version: V3.5 SP16 (actually: Lenze PLC Designer 3.23)

Thanks for this great library!

stefandreyer, 2021-07-20:
Hi Stefan,

the easyest and preferred way would be to place  the MQTT_IN_OUT outside of a GVL, may in a Programm or FB.

Else you have to move the content of the FB_EXIT methods in a new method like this exiter:

[image]

If you have fun with the lib you may leve me a star...

Have fun.

BR Stefan

stefanroelofs, 2021-07-20:
Thanks for the response. 👍 I will test this tomorrow.
Stefan

stefandreyer, 2021-07-22:
Hi Stefan

did you have succsess?

BR Stefan

stefanroelofs, 2021-07-23:
Hi, 

Unfortunately not yet.

I placed the MQTT_IN_OUT in a function_block:
[image]

And I also put the attribute in several existing 'destroy' methods:
[image]

This exception is thrown on a Warm Reset:
[image]

stefandreyer, 2021-07-25:
Hi,

I use this setup in a PRG, you may try a PRG:
[image]

BR Stefan

stefanroelofs, 2021-07-27:
Hi,

I've quickly created a small test project with the setup in a PRG like you say, and I don't have any issues anymore.

[image]

Both when I run it in simulation, or when I download it into a PLC, I don't see any exceptions anymore. 
Thanks for your assistance!
Stefan

closed 2022-11-22 by stefandreyer