stefandreyer/CODESYS-MQTT #57: I Can't publish any payload

abaubi, 2022-04-20:
Hello

I can't publish any payload. I'm using Codesys Control Win x64 device on windows 10 to test this MQTT library.

I configured MQTT Client, Subscribe and Publish. I got Client connection and Subscribe to work fine but Publish is still not working. I set Publish send input at true and sendactive and a few later sendtimeout are at true state.

I tested the example testoutoftheboxnotls and it's happening the same.

Any help?
 
Thanks
[image]
[image]

https://drive.google.com/file/d/1CoVkFo8iJUdn7TvBPwL0p82sW3KFi5M7/view?usp=sharing

stefandreyer, 2022-04-20:
Hi,

you have to use the ClientFB instance in your MQTT_IN_OUT var.

Please let me know if this works.

BR Stefan

abaubi, 2022-04-20:
Hi, Stefan thanks for your fast answer.

I'm not sure if I understant what you told me. I'm setting MQTT_IN_OUT:=PLC_PRG.MQTT_IN_OUT, that is used in MQTTClient FB, in subscribe and publish FB's as you can see in attachments captures. Subscribe is working but Publish is not working. I'm following example testoutoftheboxnotls using similar code in ladder.

I tried to compile MQTTPublish_0.SetMqttInOut(MQTT_IN_OUT:=PLC_PRG.MqttClient_0.MQTT_IN_OUT); but is compiling with errors.

I attached you the project archived if you could check it out.

https://drive.google.com/file/d/1CoVkFo8iJUdn7TvBPwL0p82sW3KFi5M7/view?usp=sharing

[image]
[image]
[image]
[image]
[image]

stefandreyer, 2022-04-21:
hi,

i mean a call like this:
[image]
let me know if this works

BR Stefan

abaubi, 2022-04-21:
Hi Stefan,

That was. It's working now,.

Thanks so much for your help and for your great job.

jshroyer031, 2022-04-21:
Stefan,

Can you send me the latest MQTT library that is functional?

Jordan

stefandreyer, 2022-04-22:
Hi jshroyer031

you can use 1.2.0.6 from this repo, its the last I developed.

BR Stefan

closed 2022-11-22 by stefandreyer