stefandreyer/CODESYS-MQTT #60: Https Rest-api call

SanderNID, 2022-12-14:
Hi Stefan,

Unfortunately I am not able to send you a direct mail.
I am trying to send POST and GET request via Codesys (https). I was looking in your libraries and found the HTTP_GET function in CommonTypesAndFunctions.
I am trying to get this to work however the OSCAT_NETWORK_TYPES.URL is poorly described, I have lots of variables which I think I do not need.
Furthermore the Authorization is hard coded on: "Authorization: Basic" whilst I need "Authorization ApiKey".
[image]

Is there a chance you can help me further or create a dedicated library like you did for MQTT?

My current goal is to send a post call via https:
url: https://api.notimportant.com/project/testproject
apikey: Authorization: ApiKey <generatedkey>

Best regards,
Sander

stefandreyer, 2022-12-19:
Hi Sander,

nice to hear from you.

I allready spend some time implementing https get for [solecast.com.au](https://solcast.com/) and getting anoying https errors. I extended the librarys for httos get. Post isn't implementet by oscat and would interesting to do. 

I would be nice if we could put our heads togehter for this.

You may want to view my extended libs?

Are you common with https?

BR Stefan

SanderNID, 2022-12-20:
Hi Stefan,

I have the basic knowledge of https, however I am in no way an expert.
Nevertheless I would be happy to help!

Am I already able to view the source code of your extended libs, so I can take a look?
Could you also explain the problem for https post you are facing currently?

Best regards,
Sander

stefandreyer, 2022-12-20:
Hi, 

I never had a close look for post because I didn't need it. But it seems not so easy to write a post FB which doesn't existis until now. You have to implement the whole post protokoll...

BR Stefan

SanderNID, 2022-12-21:
Hi,

It actually does exist, however it is a paid compiled library (https://store.codesys.com/en/iiot-libraries-sl.html).
Could you send the link to the already existing https GET? 
I will take a look into POST and hopefully figure it out.

Best regards,
Sander

stefandreyer, 2022-12-28:
Hi Sander,

I ment doesn't exist in the OSCAT library context I'm using. 

The http get from Network 1.3.5.2 already support https:
[image]

I atteched the CommonTypsAndFunctin lib containing the IP_Control which checks this:

[image]

You may have a try, but be aware to send sensitiv data by testing.

[CommonTypesAndFunctions 1.1.0.7.zip](https://github.com/stefandreyer/CODESYS-MQTT/files/10313541/CommonTypesAndFunctions.1.1.0.7.zip)

BR Stefan

closed 2025-11-01 by stefandreyer