stefandreyer/CODESYS-MQTT #63: CMPTLS error

Damian643, 2023-05-01:
Hi Stefan,

I'm trying to implement your software on a PLC using Codesys 3.5 SP17 Patch 3. I was able to fix most errors that came up by downloading some of the libraries you have provided and fixing some of the placeholders. However I can't seem to get rid of this following error when logging into the controller. 

[image]

From what I can tell online this error is related to SysSocket2 Library but I was wondering if you have any ideas or experience dealing with this error. 

Thanks!

stefandreyer, 2023-05-02:
Hi Damian

the os of your plc doesn't implement openSSL. Do you want to use TLS? 

If not, there must be a mok Libraray with dummys for the TLS implementation. You need it?

BR Stefan

Damian643, 2023-05-02:
Hi Stefan,

Thanks for the reply. I was hoping to use TLS if possible (Are you saying it would not be possible with this plc?), but am happy to just get it initially working. Yes that would be great, do you have access to it already or could you point me in the right direction? 

Thanks!

stefandreyer, 2023-05-05:
Hi,

yes, TLS will not work, contakt your PLC vendor for help. 

I didn' found the mentioned Library, you may comment out the TLS calls in the TLS sections of IP_CONTROL:
[image]

BR Stefan

closed 2023-11-29 by stefandreyer