stefandreyer/CODESYS-MQTT #64: Update der Integration.md

VenMarMic, 2023-08-30:
Sehr geehrter Herr Dreyer,

zu aller Erst: vielen Dank für Ihre Arbeit!
Ich habe zu Testzwecken eine Turck PLC bekommen, die von Haus aus kein "MQTT spricht". Daher war ich sehr froh über Ihr Projekt.
Bei der Implementierung habe ich mich an Ihren integration guide gehalten. Dieser hatte mich zu einem Funktionierendem Connect geführt, aber am publish bin ich gescheitert.
Nach etwas tüfteln bin ich dann darauf gekommen, dass der Part (BASIC HANDLER): "and an instance of SD_MQTT.HANDLE_MQTT for communication to the broker" veraltet ist.
Das HandleMQTT bzw. nun ClientMQTT ist bereits im Struct instanziiert. Auch der ADR(...) Teil kann nun weggelassen werden (Dank ihrer Implementierung).

`PROGRAM MQTTConnect
VAR
	stMQTTConnectData: MQTT.MQTT_IN_OUT:=(ClientID:='TX700Test');
//	fbMQTTClient: MQTT.MqttClient; OBSOLETE HANDLE_MQTT
END_VAR

stMQTTConnectData.clientFB(
	MQTT_IN_OUT:= stMQTTConnectData, 
	ENABLE:= TRUE, 
	URL:= 'mqtt-broker.vm.service-ventury.de:1883', 
	TIMEOUT:= T#2S, 
	TLS:= FALSE);
	
Publish();

PROGRAM Publish
VAR
	fbPublish: MQTT.MqttPublish;
	bPublish: BOOL;
	init: BOOL;
END_VAR

IF NOT init THEN
	init := TRUE;
	fbPublish.SetMqttInOut(MQTT_IN_OUT:= MQTTConnect.stMQTTConnectData);
END_IF;
//go on	

		fbPublish(
			Topic:= ADR('TestTopic/FromCodesys'), 
			PayloadString:= ADR('YES WE CAN'), 
			PublishAsString:= TRUE, 
			QoSIn:= MQTT.QoS.ExactlyOnce, 
			MRetain:= FALSE, 
			send:= bPublish);
`

Das erleichtert es eventuell anderen Neulingen

annD-annD, 2023-11-23:
@VenMarMic Danke für den Input. Ich verwende diese geniale Bibliothek von Stefan schon länger - konnte den ganzen Ablauf noch immer nicht ganz durchblicken. Ich kopiere einfach Beispielblöcke mit automatischen Topics immer weiter und verwende es zum Senden meiner Werte. Jeder Fortschritt in der Dokumentation hilft mir weiter. Ich muss mich da nochmal reindenken.

@stefandreyer Deine Bibliothek läuft übrigens auch wunderbar auf einer Wago 750-8212 (PFC200) mit aktueller Firmware 26 und Codesys 3.5.19.20 (ohne eCockpit). Es fallen somit keine Lizenzkosten für die Software mehr an.

Gruß annD

stefandreyer, 2023-11-29:
Hi,

ich habs mal aktuallisiert. Schauts euch mal an.

Grüße Stefan

closed 2025-11-01 by stefandreyer