stefandreyer/CODESYS-MQTT #65: Having trouble subscribing

DBraszak, 2024-01-04:
Hello Stefan,

Following your documentation I have succesfully Published payload into the broker and received it on the other application, but I'm having an issue with Timeout during subscribing to a topic:

[image]

[image]

[image]

[image]

I have spent some time trying to debug it, but I'm out of ideas.

I would appreciate some help! Thanks in advance.

stefandreyer, 2024-01-04:
Hi,

in the ClientFB there is a error History, is there any entry? Maybe the broker resets the connection. Which broker do you use?

Can you make a Wireshark snipped for the subscription attemp?

BR Stefan

DBraszak, 2024-01-04:
Error history looks empty:

[image]

Wireshark scan starting from CPU start:

[image]
So basically nothing happened on the network after switching on the Subscribe input.

My setup:
CODESYS Control Win V3 - x64 on my laptop

MQTT Mosquitto broker hosted on and Rapsberry PI 4b with Home Assistant OS

Subscribing from my laptop through the command line with mosquitto_sub works correctly with the same parameters.

stefandreyer, 2024-01-05:
Hi,

is your dynamic memmory big enough? Check if CreatDynamicContentFailed is set in ClientFB, if so, increase dynamic memory ammount.

Hint for this, the Pointer to the history is zero.

BR Stefan