stefandreyer/CODESYS-MQTT #66: Bug in ClientWillTopic

SvenJoedicke, 2024-04-05:
Ich habe wahrscheinlich einen Bug beim ClientWillTopic entdeckt . Ab einer gewissen Größe geht der HANDLE_MQTT in Störung und es ist nicht möglich sich mit den MQTT-Broker zu verbinden. Als Beispiel habe ich einen Gutfall und einen Schlechtfall mit exakt einem Zeichen mehr.

I probably discovered a bug with ClientWillTopic. When the HANDLE_MQTT reaches a certain size, it malfunctions and it is not possible to connect to the MQTT broker. As an example, I have a good case and a bad case with exactly one more character.

Good Case:
[image]

Bad Case:
[image]

Test-Sourcecode:
<pre><code>
// Declaration
PROGRAM PLC_PRG
VAR
	_MQTTInOut  : MQTT.MQTT_IN_OUT;
	_MQTTInfo   : MQTT.MQTT_INFO;
	_MQTTCLient : MQTT.HANDLE_MQTT;
	_ConfigDone : BOOL;
	_ErrorC     : DWORD;
	_ErrorT     : BYTE;
	UseValidMqttTopic      : BOOL := TRUE;
	LastUseValidMqttTopic  : BOOL;
	ValidMqttWillTopic     : STRING(255) := 'AAAAA/BBBBB/CCCCC/DDDDD/EEEEE/FFFFF/GGGGG/HHHHH/IIIII/123456';
	InvalidMqttWillTopic   : STRING(255) := 'AAAAA/BBBBB/CCCCC/DDDDD/EEEEE/FFFFF/GGGGG/HHHHH/IIIII/1234567';
	ChoosenClientWillTopic : STRING(255);

	Connected : REFERENCE TO BOOL := _MQTTInOut.BROKER_CONNECTED;
END_VAR 
</code></pre>
<pre><code>
// Implementation
IF _MQTTInOut.ClientWillTopic <> ChoosenClientWillTopic OR_ELSE UseValidMqttTopic <> LastUseValidMqttTopic THEN
	_MQTTCLient.ENABLE := FALSE;
	IF UseValidMqttTopic THEN
		ChoosenClientWillTopic := ValidMqttWillTopic;
	ELSE
		ChoosenClientWillTopic := InvalidMqttWillTopic;
	END_IF
	_MQTTInOut.ClientWillTopic := ChoosenClientWillTopic;
	LastUseValidMqttTopic := UseValidMqttTopic;
END_IF

_MQTTCLient(
	MQTT_IN_OUT := _MQTTInOut,
	MQTT_INFO	:= _MQTTInfo,
	URL			:= '192.168.1.20:1883',
	TIMEOUT		:= T#5S,
	TLS			:= FALSE,
	ERROR_C		=> _ErrorC,
	ERROR_T		=> _ErrorT
);
_MQTTCLient.ENABLE := TRUE;

Connected; // Only for monitoring
</code></pre>

stefandreyer, 2024-04-17:
Hi Sven,

ich denke ich weiß wo das her kommt.

Bin dran, Grüße

stefandreyer, 2024-04-21:
Hi Sven,

ich bekomme das bei mir nicht nachgestellt, kannst du mir was zum Broker und den Versionen der Libs sagen?

[image]

Bei mir bekommt er den Connect.

Hast du auch eine andere Will Message genommen?

Grüße

SvenJoedicke, 2024-04-22:
Hallo Stefan,

Tatsächlich nutze ich einen etwas älteren Broker, Version 1.6.15

Und ja, ich nutze in der richtigen Applikation eine andere größere Message,
mit demselben Problem.

Mfg
Sven