stefandreyer/CODESYS-MQTT #73: thingsboard mqtt basic

LsIntSys, 2024-12-04:
Hello Stefan.

Please explain us how we can use your library with thingsboard Mqtt basic.
Where we can put username and password?

Regards

stefandreyer, 2024-12-05:
Hi

username an password are passed in the url:
http://hans:geheim@www.example.org:80/demo/example.cgi?land=de&stadt=aa#abschnitt1
|       |      |         |         |   |                   |               |
|       |      |         Host      |   Pfad                Query           Anker
|       |      Passwort            Port
|       Benutzer
Protokoll

you may have a link to the used mqtt structure?

BR Stefan

LsIntSys, 2024-12-05:
Hi
This is the example that gives Thingsboard:

> mosquitto_pub -d -q 1 -h "YOUR_TB_HOST" -p "1883" -t "v1/devices/me/telemetry" -i "YOUR_CLIENT_ID" -u "YOUR_CLIENT_USERNAME" -P "YOUR_CLIENT_PASSWORD" -m {"temperature":25}

Can you give an example how we can call your function blocks?

BR

stefandreyer, 2024-12-09:
Hi,
did you check the implementation examples?

Are the more questions?

BR Stefan

appienvanveen, 2025-02-10:
We expereienced this as well. Looked like our port was blocked so it looked like it wasn't working