stefandreyer/CODESYS-MQTT #76: Limited String payload size

JensjeM, 2025-03-01:
Hi stefan, 

For a proof of concept project I wanted to use your library to get PLC data to a MQTT broker on a server I own. When I tested it by making a counter count up and sending it as a string to my broker everything worked fine. But when I increased the string size the pointer to that string stopped working and is giving me a "Dereference of invalid pointer" when I try to run it. I tried to increase the string sizes in the project and increased the dynamic memory allocation to 20000 but nothing changed.  The payload is a JSON string that looks like: 
{"Context":1,"Description":"Test message","ReportText":"Maintenance","Status":2,"Priority":"M"} With this being the shortest one. Could you please explain me how to solve this?

Code is as follows:

`PROGRAM MQTT_Test
VAR
	init : BOOL;
	MQTT_IN_OUT : MQTT.MQTT_IN_OUT;
	Publish : MQTT.MqttPublish;
	Send : BOOL;
	tnDelay: TON;
	Topic : STRING(255) := 'plc/data' ;								
	PayloadString : STRING(1024);
END_VAR'

`PayloadString := Create_POST.Create_PUB;	

IF NOT init THEN
Publish.SetMqttInOut(MQTT_IN_OUT);
init := TRUE;
END_IF

tnDelay(IN:= TRUE, PT:= T#5S , Q=> Send, ET=> );

MQTT_IN_OUT.clientFB(
	MQTT_IN_OUT:= MQTT_IN_OUT, 
	ENABLE:= TRUE, 
	URL:= 'Admin:Admin123!@0.0.0.0:1883',  
	TLS:= FALSE
);

Publish(
	Topic:= ADR(Topic), 
	PayloadString:= ADR(PayloadString), 
	PublishAsString:= TRUE, 
	Payload:= , 
	PayloadLength:= , 
	QoSIn:= 0, 
	MRetain:= FALSE, 
	send:= Send, 
	done=> , 
	sendActive=> , 
	SendTimeout=> , 
	CountSend=> );

IF Publish.done THEN
	Send:= FALSE;
	tnDelay(IN:= FALSE);

END_IF`

Thanks in advance!

stefandreyer, 2025-03-02:
Hi,

this should work as aspected, I'll user JSON large strings too. You may show a screenshot where the invalid pointer hits? 

BR Stefan

JensjeM, 2025-03-02:
Hi Stefan, thanks for the reply! 

As I am not close to a PLC I will get back to you first thing in the morning tomorrow.

JensjeM, 2025-03-03:
Hi Stefan, 

I retried and the problem is not "Dereference of invalid pointer" but "Value of the expression cannot be retrieved"

[image]

[image]

[image]

[image]

This is how my project looks as of now. 

I'm sorry if it is stupidly simple mistake, this is my first  time working with codesys for a project ever

stefandreyer, 2025-03-03:
Hi, the client is not connectet, please check error historie in client FB an you credtials.

The value isn't displayed because the pointer points to 80 max chars. This thinks the develompments system, in real its more an no problem.

br

JensjeM, 2025-03-03:
Hello, 

Somehow my error history is completely empty (Screenshot). I thought by calling MQTT.MQTT_IN_OUT I would automatically call on the MQTTClient FB. Since the comment said it was not to be called upon directly, and when I try to send a byte that counts up it works fine but when I try to convert it to a string it does not. 

[image]

Please let me know if you have any ideas.

Thanks in advance

stefandreyer, 2025-03-03:
Hi It still looks like the client doesn't connect, see mqtt_in_out.broker_connected

you may post me the project?

Where do you see that it works? on the MQTT Broker?

BR Stefan

JensjeM, 2025-03-09:
Hello Stefan,

Sorry for the late reply. I had the project working with longer strings on the mosquitto test broker. Now my next problem is enabling the TLS. As soon as I do that I can't seem to get it to connect to anything. Since Github doesn't allow projectarchives to be send over I can't really show you the file. With the info from the pictures above, do you have any tips on how to debug it or what I am looking for?

With kind regards,

stefandreyer, 2025-03-13:
Hi,

for investigation of TLS connection problems you have to use Wireshark and monitor tcp traffik, maybe there yopu get a hint what ist the Problem, The Libary gives only les infor if the connection is not established.

BR