stefandreyer/CODESYS-MQTT #80: MQTT Publish send timeout in Codesys V3.5.21

teejc, 2025-08-09:
Hi,

Thanks for creating this wonderful MQTT library for Codesys.

Currently I'm testing out the TestMQTTGithubWindowsWithoutTLS project and facing the problem with publishing data. I'm using your MQTT library version 1.2.0.7 and latest dependencies. I downloaded the program into my Control Win V3 X64 version 3.5.21 to test with a local mosquitto broker. 

In the mosquitto verbose mode logging i can see the codesys client is connected to the broker. By my understanding, the program should be publishing the tick string every 2 second. However i'm not seeing the publish message in the mosquitto logging. At the same time, in Codesys i notice the publish FB PayloadString value is constantly changing but sendActive and SendTimeout bits are always true. 

I did try publish some random string from MQTTX to the topic "CODESYS-MQTT-BY-SD/State" and i'm able to see the data i send in the StringCallback list, so the subscribe FB should be working.

Could you help look into this?

Thank you.

[image]

stefandreyer, 2025-08-09:
Hi,

do you have a vailid referenz for MQTT_IN_OUT here?:

[image]

teejc, 2025-08-10:
There's no value compare to yours:

[image]

This is my error history:

[image]

This is my program code:

[image]

stefandreyer, 2025-08-10:
Hi,

do you have any problems allocating dynamic memory?

is this false in your clientFB too?:

[image]

teejc, 2025-08-11:
I did allocate a certain amount of dynamic memory, as you can see below:

[image]

The bit in clientFB is False.

[image]

stefandreyer, 2025-11-01:
Hi,
still issues?